seed-solutions/aero-ros-pkg #271: Kineticでaero_startupが動いていない

hyaguchijsk, 2018-01-11:
reported from @chibi314

chibi314, 2018-01-11:
すいません、今確認したらkineticでもrvizでモデルの表示はできました。
ただ、indigoと違って最後までいってない気がします。（Everyone is happy!的なメッセージが出てない）
以下のメッセージが最後です。

```
You can start planning now!

could not open /dev/aero_lower
[ INFO] [1515661578.464140638]: starting aero_hardware_interface
[ INFO] [1515661578.464184777]:  create publisher
[ INFO] [1515661578.465569207]:  create stroke publisher
[ INFO] [1515661578.466552129]:  create error publisher
[ INFO] [1515661578.467442653]:  create in_action publisher
[ INFO] [1515661578.468396173]:  create command sub
[ INFO] [1515661578.474008132]:  create wheel servo sub
[ INFO] [1515661578.480101217]:  create wheel command sub
[ INFO] [1515661578.485610442]:  create utility servo sub
[ INFO] [1515661578.490696788]:  create interpolation service
[ INFO] [1515661578.491803885]:  create send joints service
[ INFO] [1515661578.492836324]:  create get joints service
[ INFO] [1515661578.493944468]:  create grasp control service
[ INFO] [1515661578.495262341]:  create status reset sub
[ INFO] [1515661578.500713879]:  create timer sub
[ INFO] [1515661578.500863336]:  done
[ INFO] [1515661578.500942185]: configuring done
```

Horisu, 2018-01-11:
https://github.com/ros-planning/moveit_ros/issues/454 曰く，Everyone is happy!はみんなおかしいと感じていたようで，jade以降はYou can start planning now!のみになっています．

chibi314, 2018-01-11:
@Horisu なるほど、そこは問題ないのですね　ありがとうございます。

chibi314, 2018-01-11:
それ以外に、最初のあたりでxacroに関するwarningみたいのが出るのと、以下が出ます。

```
/opt/ros/kinetic/lib/moveit_ros_move_group/move_group: error while loading shared libraries: libmoveit_planning_scene_monitor.so.0.9.7: cannot open shared object file: No such file or directory
process[move_group-14]: started with pid [32481]
process[wait_interpolation_manager-15]: started with pid [32495]
[move_group-14] process has died [pid 32481, exit code 127, cmd /opt/ros/kinetic/lib/moveit_ros_move_group/move_group __name:=move_group __log:=/home/anzai/.ros/log/af5f198c-f6b1-11e7-95d7-c85b7656499d/move_group-14.log].
log file: /home/anzai/.ros/log/af5f198c-f6b1-11e7-95d7-c85b7656499d/move_group-14*.log
```
これは大丈夫でしょうか？

Horisu, 2018-01-11:
move_groupは落ちないで欲しいノードです．
同じエラーがkineticのチュートリアルレベルで出ているという質問もあるのでmoveit自体の問題かも知れません． https://answers.ros.org/question/278847/problem-in-moveit-tutorials-step-movegroup-interface-tutorial/

chibi314, 2018-01-11:
moveitのチュートリアルを行ってもやはり同じように失敗しました。
`roslaunch aero_moveit_config move_group.launch`を行っても同じエラーが出ます。原因は以下のようです。

```
/opt/ros/kinetic/lib/moveit_ros_move_group/move_group: error while loading shared libraries: libmoveit_planning_scene_monitor.so.0.9.7: cannot open shared object file: No such file or directory
```

/opt/ros/kinetic/lib/libmoveit_planning_scene_monitor.so.0.9.11はありました。バージョンが違っている？

chibi314, 2018-01-11:
`sudo apt-get install ros-kinetic-moveit*'としてmoveit関連を最新にしたところ解決しました。

closed 2018-01-12 by chibi314