seed-solutions/aero-ros-pkg #276: set_positionにかかる時間が関節数に依存している？

Horisu, 2018-02-07:
上半身のハンド以外関節を同時にを送ると，set_position前後で20msかかります．
送る関節数を減らすと，なぜかこの時間が増えているようです．

timeでset_position前後を測る
関節１つ:50ms(インターポレーションを割ると，49msに見える)
関節2つ：46~47ms
関節5つ:42ms

Horisu, 2018-02-07:
どれもseedコントローラに送る命令長は変わっておらず，中身の0x7fffの数が変わっているだけのはずです．

sasabot, 2018-02-07:
該当行を特定しました。結果、マイコンのバグと思われます。

0x7fff でマスキングする関節数を増やすと、指令からの返りをreadするときに関節一個につき数msのオーバーヘッドが入るようです。
１関節だけしか送らない場合は最大30msのオーバーヘッドが入ります。

なお、コントローラ側のコードでは、AeroControllerProto.ccのSEED485Controllerのread関数の中のser_.read_someが該当行ですが、これはマイコンから読む以外何もしていないはずです。

sasabot, 2018-03-23:
新ファームでもこのバグは直っていないようです。

sasabot, 2018-03-23:
ファームの中のソースコードを読んだ推測として、

０x７FFFのときは指令を送っていないけど、送っている送っていないに関わらずREAD待ちをしている（なので値が返ってこずタイムアウト待ちで時間加算）が原因？

SEED_SCM.c L597
```
        timer = SEED_Timer;
        while(SEED_CAN1Buf() == 0){
                if(timer + time_out < SEED_Timer) break;
        }
```

SEED_AERO.c L2191
```
               else if (TRG[snd_num] != 0x7FFF ){ 
```
に対し、elseがないので値が送られていないと思われる

@y-shiigi
ということで、検証・検討して頂けないでしょう

y-shiigi, 2018-03-25:
@sasabot 
了解しました。該当行をチェックします。
ちなみに、そちらで検証しているテスト方法を教えてもらえませんか？

sasabot, 2018-03-25:
@YoheiKakiuchi 
最小限のテストコードの共有お願い致します。

テストコードがなくても、AeroControllerProto.ccで
```
#include <chrono>
```
して、SEED485Controllerのread関数のread_some前後に
```
auto start = std::chrono::high_resolution_clock::now();
size_read = ser_.read_some(buffer(read_buffer, RAW_DATA_LENGTH));
std::cout << "time: " << std::chrono::duration_cast<std::chrono::milliseconds>(std::chrono::high_resolution_clock::now() - start).count() << "ms" << std::endl;
```
を足せば、aero_bringupで指令を送ったときに何msかかっているか表示されると思います。

これで例えばコマンドラインから
```
rostopic pub /aero_controller/command trajectory_msgs/JointTrajectory "header:
  seq: 0
  stamp:
    secs: 0
    nsecs: 0
  frame_id: ''
joint_names:
- 'r_wrist_y_joint'
points:
- positions: [0.5]
  velocities: [0]
  accelerations: [0]
  effort: [0]
  time_from_start: {secs: 1, nsecs: 0}" 
```
などと送って、10msが返ってくるのが理想で、30ms～50msが返ってきた場合は何かがおかしいです

y-shiigi, 2018-03-25:
現象再現できました。ありがとうございます。（38[ms]と出てます）
デバッグ完了次第、現地でアップデートするかマイコン送付で対応致します。

y-shiigi, 2018-03-25:
>これで例えばコマンドラインから~などと送って、10msが返ってくるのが理想で、30ms～50msが返ってきた場合は何かがおかしいです  

コマンドを送る前後でtimeの値は変わらなくなったので、本Issueのバグは直ったと思います。
ただ、指令PCによってばらつきがあって、ノートPCでは23[ms]で安定していて、Brixでは7～8[ms]でたまに0[ms]が出力されます。(#82 に関連して、使用するBrixによって通信周期がバラついていたので、その辺な気もしますが）

ちなみに、バグ検証を兼ねて``ik_sample_node``,``fk_sample_node``を実行したら、rvizにリアルタイムに反映されなくなりました。 #306 の変更の影響？前からでしたっけ？

sasabot, 2018-03-25:
対応ありがとうございます。

> コマンドを送る前後でtimeの値は変わらなくなったので、

これは送る関節数を変えても(0x7fffのマスキング数を変えても)変わらなくなったということでしょうか。
具体的にどういう変更を加えたかの報告もお願いできますでしょうか。

rvizに反映されなくなったのはこちらでも確認されている現象です。前はぴくぴくしていたということは少なくとも値を取ってきて流す処理は以前はされていて、ぴくぴく防止を入れた影響はあると思われます。調査中です。