seed-solutions/aero-ros-pkg #279: eusモデル

hyaguchijsk, 2018-02-14:
現状整備されていません．
3月中にはなんとかします．

### これまでの経緯

OpenHRPのvrmlモデルを作っておいてeusに変換し，
rtm-ros-robot-interfaceを継承するというのをやってきていましたが，
wrlを整備すること自体にひと手間あり，やめています．

現状では各部品が簡単に交換できるメリットをとって，
部品ごとに作ったURDFを合体させて一つのURDFを作る方式にしています．

### 基本方針

合体システムはおそらく，eusでも同様に
各部品ごとにeusモデルを作っておいて，
接続部でassocすれば実現できるはずです．

hyaguchijsk, 2018-02-14:
[image]

全身モデルからのeusモデル生成方法

```
$ roscd aero_description/robots
$ rosrun xacro xacro.py aero.urdf.xacro > aero.urdf
$ rosrun collada_urdf urdf_to_collada aero.urdf aero.dae
$ rosrun euscollada collada2eus_urdfmodel -I aero.dae -O aero.l -C aero.yaml
```

yamlは以下のようなものを用意する必要あり．

```
rarm:
  - r_shoulder_p_joint : rarm-shoulder-p
  - r_shoulder_r_joint : rarm-shoulder-r
  - r_shoulder_y_joint : rarm-shoulder-y
  - r_elbow_joint      : rarm-elbow-p
  - r_wrist_y_joint    : rarm-wrist-y
  - r_wrist_p_joint    : rarm-wrist-p
  - r_wrist_r_joint    : rarm-wrist-r
larm:
  - l_shoulder_p_joint : larm-shoulder-p
  - l_shoulder_r_joint : larm-shoulder-r
  - l_shoulder_y_joint : larm-shoulder-y
  - l_elbow_joint      : larm-elbow-p
  - l_wrist_y_joint    : larm-wrist-y
  - l_wrist_p_joint    : larm-wrist-p
  - l_wrist_r_joint    : larm-wrist-r
torso:
  - waist_y_joint : torso-waist-y
  - waist_p_joint : torso-waist-p
  - waist_r_joint : torso-waist-r
head:
  - neck_y_joint : head-neck-y
  - neck_p_joint : head-neck-p
  - neck_r_joint : head-neck-r
### leg
#
#
#

##
## end-coords
##
torso-end-coords:
  translate : [-0.032, 0.0, -0.3507]
head-end-coords:
  translate : [0.079, 0.0, 0.1035]
  rotate    : [0.0, 1.0, 0.0, 99.9999]
rarm-end-coords:
  translate : [0.0, 0.0169, -0.174]
  rotate    : [0.0, 1.0, 0.0, 90.0]
larm-end-coords:
  translate : [0.0, -0.0169, -0.174]
  rotate    : [0.0, 1.0, 0.0, 90.0]

##
## reset-pose
##
angle-vector:
  reset-pose : [0.0, 0.0, 0.0,
                0.0, 0.0, 0.0,
                0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0,
                0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0 ]
```

YoheiKakiuchi, 2018-02-14:
目標 pr2eus に準拠する
https://github.com/jsk-ros-pkg/jsk_pr2eus/pr2eus

作るもの
aero.l (ロボットモデル、型番が付いてもいい、自動変換されるファイル)
  aero-interface.l (rosとのインターフェース、robot-interface クラスを継承、pr2-interface.lを参照)
  aero-utils.l (aero-interface.lからロードされる。ロボットモデルの泥臭いHotFixなどが書いてある。pr2-utils.lを参照)

モデルを作る場合もaero.lを直接ロードしないで、aero-interface.lをロードして作る。
(pr2jsk, hrpsys_ros_bridge_tutorialsのロボットはすべてこの方式)

YoheiKakiuchi, 2018-02-14:
モデル変換は、urdf -> dae -> euslisp の場合
~~~
$ rosrun collada_urdf urdf_to_collada aero.urdf aero.dae
$ rosrun euscollada collada2eus aero.dae aero.l aero.yaml
~~~

urdf -> euslisp の場合
~~~
$ rosrun euscollada collada2eus_urdfmodel -I aero.urdf -O aero.l -C aero.yaml
~~~

センサ情報がついていない場合(urdfオリジナルの場合は、センサ情報は基本つけられない)は、後者で問題ないように思うのだがどうだろうか？

hyaguchijsk, 2018-02-14:
後者は以下のようになり実行できません．

```
$ rosrun euscollada collada2eus_urdfmodel -I aero.urdf -O aero.l -C aero.yaml
;; Input file is: aero.urdf
;; Config file is: aero.yaml
;; Output file is: aero.l
torso@17
larm@9
rarm@1
head@21
Warning: The DOM was unable to create an element named robot at line 10. Probably a schema violation.

Error: Failed to load file:/home/leus/ros/indigo/src/aero-ros-pkg/aero_description/robots/aero.urdf

[ WARN] [1518586511.184539907]: This file (aero.urdf) is not collada file.
```

YoheiKakiuchi, 2018-02-14:
> 後者は以下のようになり実行できません．

手元では、メッセージは同じメッセージがでるけど、それでもaero.lできていないかな？

hyaguchijsk, 2018-02-14:
確かに．．．
メッセージがわかりにくすぎますね．

[image]

hyaguchijsk, 2018-02-14:
robot-interfaceにした場合，
今はFollowJointTrajectoryActionを使っているため，
#281 の議論を進めないといけませんね．

YoheiKakiuchi, 2018-04-04:
aeroeus moves to https://github.com/jsk-ros-pkg/jsk_robot/pull/906

closed 2018-04-04 by YoheiKakiuchi