seed-solutions/aero-ros-pkg #280: abort trajectory collisionが出て動かなくなる

Horisu, 2018-02-14:
``` bash
[ INFO] [1518588701.930505940]: upper joint trajectory thread: starting!                                                │················································
[ERROR] [1518588701.950568612]: upper joint trajectory thread: abort trajectory collision!                              │················································
[ INFO] [1518588701.950621689]: finished upper joint trajectory thread 20.000000                                        │················································
[ INFO] [1518588704.826961313]: finished lower joint trajectory thread 2896.000000
```
前までは動いていたコードが動かなくなっています．
ピックアンドプレースをしていたのですが，掴んだ後持ち上げる動作が実行されません．
手元のノード側ではエラーは出ておらず，コントローラー側に以上のエラーが出ていました．

sasabot, 2018-02-14:
これはこの前(12月末)脱調検知をオンにしたのが原因ですね。
脱調するとソフトウェア側でストップがかかる、という行があって、そこで止められている挙動です。

動きが早すぎるか、ハードウェアガタが来ているか、ですが、
ソフトウェアストップがない方がいいというのであれば該当行を抜くことも出来ます。

どうしたいですか?

Horisu, 2018-02-14:
遅くすれば動くようになりました．
止まった後当分動かないので何もできなくなるのが辛いです．

sasabot, 2018-02-14:
止まったあと当分動かないのは確かに問題ですね。。。
かといってコントローラ内で脱調を検知したらコントローラ側でステータスをリセットするようにすると、今度は外部で脱調情報が欲しいときに取れなくなる。

早い話がマイコン側で、脱調フラグのリセットを早める(今10秒でしたっけ? 長すぎますね)のが正解な気もしますが、脱調検知を無視するモードがコントローラ側にあってもいいかもしれませんね。

必要であれば「脱調フラグ出ていてもいいから動けモード」を足しますが、どうしますか?

Horisu, 2018-02-14:
そうですね．それがありがたいです．

sasabot, 2018-02-14:
#282 で対処しました.

closed 2018-05-18 by sasabot