seed-solutions/aero-ros-pkg #281: ROSコマンドの送り方について

YoheiKakiuchi, 2018-02-14:
ロボットへのコマンドについて、上半身(?)は基本的に以下のソースに書いてあるノードに、設定してあるものと思います。
https://github.com/seed-solutions/aero-ros-pkg/blob/master/aero_startup/aero_hardware_interface/AeroControllerNode.cc

trajectory_msgs/JointTrajectory をメッセージで受け取るようになっているように見えます。
動作中かどうかは、in_actionメッセージを出している？ ということで合っているでしょうか？

control_msgs/FollowJointTrajectoryAction ( http://wiki.ros.org/control_msgs ) にしていない理由はなんでしょうか？
参照にした実装、ロボットがあるのでしょうか？
また、動作が終わる前に新たな指令が来た場合どう動くようになっているでしょうか？

Horisu, 2018-02-14:
現在の指令の送り方は，おっしゃる通りです．
指令が途中で来ると上書かれるようになっています．

topic経由にした経緯は別として，現在はFollowJointTrajectoryActionにしていない理由は特に無いです．
構成変化に合わせて/arm_controllerの扱う関節などを変更する方法がよくわかっていなかったなどの理由もありますが，rosparamから取ってこれる状態なので実装しようと思えばできるはずです．

sasabot, 2018-02-14:
正直FollowJointTrajectoryにして頂きたいところです。。。

補足になりますが、
動作を途中でキャンセルすることも機能としてはあります。また、一旦止めて再開もコントローラレベルで実装自体はあります(まだpreq出していませんが)。

動作が終わる前に新たな指令が来た場合についても補足すると、
例えば腕と頭を同時に動かす指令を送った後に頭だけ動かす指令を送ると、腕はそのまま前の指令のまま動いて頭だけ上書きされます。

現状のin_actionには落とし穴があって、指令レベルで管理しているのではなく、単に動いている部位があるかどうかチェックしかしていないので、腕を動かしたあとに頭に指令を送り続けると、例え腕が動き終わっていても、頭が動き続ける限り終了判定が返ってきません。

hyaguchijsk, 2018-02-14:
現状のfollowjointtrajectory周り，先にドキュメントなり解説がほしいです．

ダミーモード起動してtypefだとtorso,head,rarm,larmが存在することはわかりました．

検索すると，actionserverは以下で立てている．
https://github.com/seed-solutions/aero-ros-pkg/blob/4c693322899c294505c2d0e89ced4fc77ace3c29/aero_moveit_config/scripts/joint_trajectory_action_server.py

launchは以下．
https://github.com/seed-solutions/aero-ros-pkg/blob/4c693322899c294505c2d0e89ced4fc77ace3c29/aero_moveit_config/launch/joint_trajectory_action_server.launch

Horisu, 2018-02-14:
現状ではmoveit 対応を始めた際に作ったscriptが今も走っています.
中身はactionを受け取ってtopic を出す変換器になっています。

当時コントローラー自体のコードがよくわかっていなかったのとtopicベースになってメンテナンスが入らなくなった結果そのままの形で残っていますが、本格的に対応するならaero_controllerに内包したいです。

hyaguchijsk, 2018-02-14:
あと，経緯については
`(send *ri* :angle-vector)`がjointtrajectoryメッセージだった時があって，
その頃の名残がずっと残っているだけで，
現状そっちも変わってしまっているので，
根本的にfollowjointtrajectoryに変えてしまうのでいいと思います．

sasabot, 2018-02-14:
既にお察しかもしれませんが、FollowJointTrajectory対応されてこなかったというのは、対応したくてもするのにそれなりの開発コストがあるからです。

まず、最大の壁は20ms問題です。一回指令を送るにしても一回指令を受け取るにしても最低20ms、合計40msかかります。

なので、どのみち最大で50ms程度の周期でしか値は送れません。(ただし、フィードバックとして送った指令をそのまま返すだけで良いなら20ms周期で回せます)

次に、現状部位ごとにスレッドを立てて各スレッドから直接コントローラに値を送る方式になっていますが、これだと最大動かす部位数×20ms+例のバグのオーバーヘッドmsだけかかってしまいます。
なので、各スレッドは特定関節の値だけ更新して、別の周期的に回っているスレッドが(一個以上スレッドが立っている場合に限り)一括して指令を送る方式に変更する必要があります。

なお、コントローラレベルでは特にムーブグループは定義されていなくて、既に上がっているスレッドに上書きしたい関節がある場合は該当スレッドを一旦殺して、関節グループを再分配する、ということをやっています。
なので、上半身全体グループに指令を送った後、左腕グループを上書きした場合、左腕指令は上書きされますが、前の全身指令も他の部位に対しては生き続けます。
もし、グループを定義するならここらへんの仕様をどうしたいかちゃんと決める必要があります。

sasabot, 2018-02-14:
あと下半身指令をどう扱うかも考える必要があります。
現状、下半身コントローラはxz移動量を受け取って角度指令に変換したものを、上記の全体(上半身)コントローラに渡しています。
軌道を送る場合は、確か下半身コントローラから変換結果だけ受け取って、受け取った結果をまとめて全体コントローラに直接送る(下半身コントローラを経由しない)とかやっていた気がします。

hyaguchijsk, 2018-02-14:
今がそのコストをかける時です．

ハードウェアの限界はそうとして，
まず現状fullbodyに当たるものがなく，
そこをまず作って，全身指令を一括で送る．

そのfullbodyに対して，部位毎のサーバがあって，
それらはfullbodyに上書きを依頼する，

という形にはならないでしょうか．

hyaguchijsk, 2018-02-14:
下半身は問題がごっちゃになっているので一度切り分けます．
上半身だけを先に考えたい．

YoheiKakiuchi, 2018-02-14:
ここらへんのseedの限界が分かっていません。
泥臭い部分とROSに相当する上位インタフェースの部分をある程度分離したいです。
seedがどういう挙動をするのか正確に分かっていないと、どういうシステムがいいか
判断するのは難しそうです。
しかもその部分は、コントローラの配線（どうやってチェインするか）やPCとどう接続するのか、
などにも依存しそうに思います。
デモがあると思うので、デモが終わり次第できるだけ早く現状がどうなっているか教えて欲しいと思います。

あまり知らないかもしれませんが、HRP2,JAXON等はshared_memoryを挟む形式で、
上位のコントローラ(hrpsys)と下位のハードウェアに依存したコントローラを分離開発
できるようになっていて、下位の速度が遅いなら遅いなりに動かせています。
（その方法が絶対的な正解だとは言いませんが）

sasabot, 2018-02-14:
> そのfullbodyに対して，部位毎のサーバがあって，
それらはfullbodyに上書きを依頼する，

のやり方を説明したのがさっきの話になります。

ただ、もともとの設計思想の名残でムーブグループが変わり得る、ロボットに翼が生えるかもしれない、腕が四本になるかもしれない、ということで、部位毎のサーバはコントローラには定義されていなくて、サーバは指令が来た段階で初めてあるグループのサーバとして定義される。という方式になっています。
なので、グループトピックを受け取ってコントローラと通信する層（サーバ層？）と、コントローラ層を分けて、コントローラ層にはサーバ層からグループ名を渡すとか、そんな方式になる気がしますが、そうでないならAeroControllerNode自体を構成に合わせて自動生成するか、腕は絶対に二本で翼は生えない、と固定するか、といった何パターンか手段が考えられます。

sasabot, 2018-02-14:
少なくとも、このロボットの場合でいうコントローラは制御担当ではなく、あくまで、角度指令をストローク値に変換し、任意に定義された部位ごとにサーバを立てて管理する、のが主な役割です。これを泥臭い部分と呼ぶか上位インタフェースと呼ぶかはよくわかりませんが、現状

seedとの通信コマンド、
ROSからのトピックに対しseedに送るコマンドを生成、

という感じで切り分けられています。
コントローラの配線については、現状、上半身seedと下半身seedそれぞれがひとつのPCに接続されている、という前提のコードになっています。関節数の変更などは設定ファイルを変えるだけで全部対応できるように既になっているので、特にコード側で何か特別なことはやっていません。
ただし、上半身seedが二つに分かれるなどのようなことがあると、今のコードは対応できません。
ここらへんについては、そもそも上半身と下半身でseedが行う内容が違いすぎて、対称性がないとかいろいろと複雑な事情があります。

sasabot, 2018-02-14:
ということで、まとめると、

- 部位変形可能という当初のコンセプトが話を複雑にしている原因のひとつ。
- 部位ごとにサーバを立てることは既にやっていて、あとは、サーバの役割が指令を送るか値を更新するかの違い。
- ただし、このサーバというのは部位ごとのトピック名には対応していので、部位トピックと別途対応させた何かをかます必要がある。
- seedが影響してくるのは20ms限界と、少ない関節数を送ると謎のオーバーヘッドが発生するというバグだけ。
- コントローラの配線とかの泥臭い部分は設定ファイルが対処してくれる。
- 関節数変更は設定ファイルでどうにでもなるけどマイコン数が変わったりするとどうしようもない。

hyaguchijsk, 2018-02-14:
基本設計としてAeroControllerProtoは上半身下半身関係なく作ってあって，
そこから人型のUpperと四足のLowerに派生させてあり，
かつこれはROSと関係なくまた包含関係としてROS層とスレッドで分けるという思想をもって作っており，
shared_memを使ってプロセスごと分けるというほどではないですが，
分離はしてあります．プロセス間通信かスレッド間通信かの違いしかないはずです．

- fullbodyはupper,lowerで定義し，これが目標値を統括して指令を行う．
- 部位毎のサーバは値を更新するのみとして，実際の指令はfullbodyで行う．
- グループの定義は静的にコードを出力するよりも設定ファイルを読んで動的に行わせたい．

という方針では，どうでしょうか．

sasabot, 2018-02-14:
subscriberをvectorで可変にして、コールバック関数をバインドして、各コールバック関数が決まったグループ名のスレッド（途中経過や終了はスレッドが返す）を立てれば行けそうな気がしないでもないですね。

ただ、他に決めるべき方針として、
- 途中の角度フィードバックは送った値をそのまま返すようにするか、それとも毎回ちゃんと20msかけてでも実際の値を取ってくるようにするか。
- グループの中に別の小グループが含まれる場合、小グループをあとから上書きした場合の挙動をどうするか。（普通のロボットはどうしているのでしょう？）

というのがあります。

YoheiKakiuchi, 2018-02-14:
seedの動作がわかっていないのですが、
上半身の全軸に関節角度指令をおくって、全軸の関節現在値を読むとどれぐらい時間がかかるの？

sasabot, 2018-02-14:
基本的にマイコンとの通信はコマンド＋数値配列で、ひとつのコマンドにつき20ms。書き込みと読み込みは別々のコマンドなのでそれぞれ20ms。単純に倍になります。

YoheiKakiuchi, 2018-02-14:
関節角度はサーボが入っていない状態でも、正しい値が読めるのかな？

あと送る指令は関節角度のみ？ それとも、関節角度と到達時間のようなものを与える？

いずれにしても、全軸のsendとreceiveを40msと割りきって、それだけやる最下層と、
それへの指令をつくるROSの層で分ければいいんじゃない。
そうなってるようにも見えるけど。

ROSの層はPR2式にするなら、グループは重ならずに常に同じ割り当てになっている。

sasabot, 2018-02-14:
グループはmoveitのIKを解く範囲で決まると思っていましたが、そういうわけではない？
torsoは腰３軸だけで良くて、腕を含まなくても、腰付きIKと腕IKが分けられるという認識で合っていますでしょうか。
PR2は腕が長いので腕だけでも結構解ける範囲は広いかもしれませんが、このロボットの場合、腰を使わないとかなり絶望的です。

YoheiKakiuchi, 2018-02-14:
trajectory_actionのグループは、rarm, larm, torso, head, base みたいになっていて、
moveitのIKは、rarm + torsoみたいにそれの組み合わせになる（ように作る）。
moveitからは、rarmとtorsoのグループにactionが送られる。

あまりmoveitで想定されているとは思えないが、rarm + torsoのactionの動作が終わらないうちに、
larm + torsoのactionが送られれば、rarmはそのまま動き続け、torsoは動作が上書きされ、larmは動作を始める。
ROSのコントローラ上では少なくともactionのグループは非同期に動く。

この場合、動作のグループとIKのグループの関係は、動作のグループの集合(セット)がIKのグループになることで、
動作側のノードは上位がどうするかを意識すること無く、指令通りに動けばいい。

sasabot, 2018-02-14:
なるほど。そういう意味では多分現状のコントローラがそういう仕様になっているので、設定を読み込んでグループのトピックの受け手を生成するようにして、スレッドの役割を指令更新方式にさえすれば、FollowJointTrajectory対応は出来そうですね。

上半身は多分これで解決で、あとはリフターをどうするかです。現状、上半身も下半身もまとめてひとつのトピックで受け取れるようになっているので、これを分けるか、どうにかする必要があります。どのみち内部処理的には下半身は下半身で専用のスレッドを立てているので、アクション化するのであれば素直に分けちゃうでいい気がします。

あと、今の仕様では下半身の指令だけは上書き出来ないようになっているので、必要であればこれも上書きできるようにしちゃうのがいいかもしれません。（上書き可能にしていなかった理由は #205 今の仕様だと問題になっていた点の一部はクリアされている気がしますが、小刻みな指令を送って大丈夫かは念のため確認しておいた方がいいかもしれません）

hyaguchijsk, 2018-02-15:
上半身の中，下半身の中は固定長で全軸送る必要があったので分けれられてませんが
上半身と下半身は分けていいはずです．

段階的に取り組める気がしてきました．
先に上半身だけでいいので
理解できたところから修正をかけていきたいです．
まずはmovegroupの整理とcontrollerにfollow~~~を移すところでしょうか．

YoheiKakiuchi, 2018-04-04:
version 2.0.0 で対応しました。

closed 2018-04-04 by YoheiKakiuchi