seed-solutions/aero-ros-pkg #283: リフターの取り扱いについて

hyaguchijsk, 2018-02-14:
問題が複合的になっているので別途考えたいです．

### そもそもの問題

リフターはxz二自由度の平行リンク機構．

### 現状

ちょっと怪しいので補足訂正をおねがいします．

urdfのロボットモデルでは，3つの回転関節としておき，
コントローラからも関節角度が返ってくる．
rvizには見た目上反映される．

一方，xzを扱うためのvirtual_lifterというのが存在する．
単純な上下と上下前後を含めたモデルが存在する？

指令値はサービスで与えている．
その位置に行けるかどうかはサービス側でIKを解いていて
サービスに問い合わせないと，つまりモデルだけでは可動域がわからない．

### やりたいこと

モデル自体を指令値に則したxz直動にしたい．
モデル自体に可動域リミットをかけて，わざわざサービスや他の言語のIFで計算させないようにしたい．
xz -> 実機への指令値はモデルもしくはコントローラで解決を試みたい．

### 問題になりそうなこと

リフターのアームの姿勢がとれなくなって衝突を引き起こしたりしないか．
干渉以外の物理的な問題，例えば重心の位置などで経験的にリミッターかけてるところがないか．

これらの問題が解決できないからと上位層のアプリケーションで解決してしまい，
結果他のアプリケーションで再利用できないものができたりしないか．

hyaguchijsk, 2018-02-14:
すごく個人的な直感なので，そうはうまくいかないというのがあれば教えてください．

これらは主にロボットモデルの問題であって，
少なくともロボットを動作させるときは直動関節のモデルとして，
モデル側で適切な可動域リミットがかかり，
かつユークリッド空間から関節指令値空間への写像が取れれば
それでうまくいくように感じています．

YoheiKakiuchi, 2018-02-14:
ローレベルの知りたいことは、ロボットへは(seedコントローラへは）どういう指令を出しているのでしょうか？
関節角度 or xz位置 ？

また、ユーザーが使うインタフェースとしては、xz位置を与えるものがあるのでしょうか？
それのソースはどこでしょうか？

図面を見たわけでないので不正確ですが、urdfから読み取れるところは、
駆動関節は2つ（上側、下側）でそれぞれパラレルリンクを駆動していて、
rviz的な見た目はmimic jointの指定をしていてできている。
パラレルリンクなので、どういう関節角度を送っても上体の水平は保たれて、xzのみが動く。

Horisu, 2018-02-14:
ユークリッド空間での処理をするにあたって、xとzの可動域リミットをどう表現するかが最大の問題になっています。

角度空間ではリフターの2関節それぞれ0度から90度で済みますが、ユークリッド空間内ではxzのリミットが分離出来ずイチョウの葉のような形になっています。

sasabot, 2018-02-14:
> ローレベルの知りたいことは、ロボットへは(seedコントローラへは）どういう指令を出しているのでしょうか？
関節角度 or xz位置 ？

残念ながらどちらでもなく、uint16_tのストローク値（モータ指令値）です。

> また、ユーザーが使うインタフェースとしては、xz位置を与えるものがあるのでしょうか？
それのソースはどこでしょうか？

コマンドラインから/aero_torso_controllerに指令を送ると、（初期の伸びた状態を0, 0とした座標系での）xz位置を与えることができます。受け取る側のソースはAeroTorsoController.ccに書いてあります。（ビルドしたかどうかで場所が変わるのでlocateしてください）

コード的に普段どういうレベルのコマンドを記述しているか知りたいのであれば、aero_std/include/AeroMoveitInterface.hhが全てです。

sasabot, 2018-02-14:
というかまず、#64 を読みましょう。

YoheiKakiuchi, 2018-02-14:
>ユークリッド空間での処理をするにあたって、xとzの可動域リミットをどう表現するかが最大の問題になっています。
> 角度空間ではリフターの2関節それぞれ0度から90度で済みますが、ユークリッド空間内ではxzのリミットが分離出来ずイチョウの葉のような形になっています。

どこかにも書きましたが、eusとそれ以外でできることが少し違います。

eusならxz可変の直動関節の定義は簡単にできるだろうと思います。mim-max-tableを使う。
また、角度空間を使ってikを解くのもそれなりに出来るはずです。(詳細はオープンな場所にないが、経験済み)

moveitをどうするかはだいぶ難しい。 pluginを書いていくことになるだろうけど。。。

YoheiKakiuchi, 2018-02-14:
> というかまず、#64 を読みましょう。

これ、結果だけまとまってないのか？

hyaguchijsk, 2018-02-14:
> eusならxz可変の直動関節の定義は簡単にできるだろうと思います。mim-max-tableを使う。

これはそうなのですが，

> これらの問題が解決できないからと上位層のアプリケーションで解決してしまい，
> 結果他のアプリケーションで再利用できないものができたりしないか．

これに該当します．
結局今もサービス呼ばないと解けるかどうかわからない．
一番下位の問題としてモデルで解決したいです．
一番欲しいのはURDFにmin-max-tableを埋め込みたい．

YoheiKakiuchi, 2018-02-14:
上位側の問題であるのは、そうなのでは。
というのは、下位側（seedやそれに送るレベル）では、xzでもらっても、関節角度でもらっても、その指令には従える。
しかし、xzの場合、送る側が不整な(その値になれない)値を送ってくる可能性があって、その場合は、
コントローラとしては、動かない、最も近いところまで動く、他、対応はできるが、
上位側の要望には一致しないので、上位の側で計画と違ってしまう。

この場合の上位は、eusやmoveit (これらは同じくらいのレイヤだよね)なので、
min-max-tableは、現状あるように見えるので、moveitでそれを反映したIKが解けるようになったらOKなのか？

hyaguchijsk, 2018-02-14:
> しかし、xzの場合、送る側が不整な(その値になれない)値を送ってくる可能性があって、その場合は、
> コントローラとしては、動かない、最も近いところまで動く、他、対応はできるが、
> 上位側の要望には一致しないので、上位の側で計画と違ってしまう。

この時に，上位側がリミットを知る手段がサービス経由というのが一度問題になり，
上位側が計画を立てるのに必要なのにオフラインでわからないということがありました．

関節角度であればリミットはモデルが知っています．
ユークリッド空間でもモデルが知っていて然るべきと思います．

eusで解決してしまうと，その情報はeusしか知らないので，
moveitや，あるいはpythonベースとかで作っているユーザーにはわからないままです．

> min-max-tableは、現状あるように見えるので、
これはURDFの表現としてできるということでしょうか？

YoheiKakiuchi, 2018-02-14:
> これはURDFの表現としてできるということでしょうか？

できない。ROS公式の方法はないので、moveitのpluginを書くのが最短に思う。
リフタだけなら、xyと関節角度の相互変換はcでもeusでも数行で書ける程度の話だよね？
モデルからの自動変換は難しいと思うので、自動変換は諦めて、書き下すのだと思う。
（もう２タイプぐらいリフタが増えたら自動変換も考えますよということで。。。）

hyaguchijsk, 2018-02-15:
別々の言語で数行でかける，の中に数式が一つでも入ると可搬性はないと考えていて，
(eusは修正しました，Cとpythonは直しませんでした，あるいはその逆，はすぐ起きる)
理想なのはmin-max-tableはどの言語からも読めるデータ構造で書き下し，
どう解釈するかはこちらで提供する，ということだと思います．
これならその数行は単にファイル読むだけになって式の書き換えは起きません．

というのと，現状やっているように，
関節角度ベースのモデルとxzベースのモデルは両方ないと，
やはりアームの姿勢がわからず衝突回避がやりづらくなりそうな気がします．
モデルを２つ用意して相互変換ができていれば問題ないはずです．