seed-solutions/aero-ros-pkg #292: githubのブランチ

YoheiKakiuchi, 2018-03-06:
githubにbranchが多くあるのですが、必要なもの以外を消したいと思うのですがいかがでしょうか？

必要性がある場合は、READMEなどにブランチの説明を書いておきたいと思います。

hyaguchijsk, 2018-03-07:
これに関してですが，基本的にbranchは使わず，
リリースタイミングでtagをうつ，

またbranchではmasterとdevelopを用意して，
developにするのはどうか，という提案を受けています．

branchに関しては，マージされたPRのものばかりと思いますので消していきますが，
0.3.0はおそらくtagに移したほうがいいものだと思います．

YoheiKakiuchi, 2018-04-11:
Branch 'master' should be changed by merging PR.
All user, admins are required to make PR, even if they have write permission.

Branch 'development' is for testing new feature.
Current status of 'development' should be discussed at issue.

A tag will be added when released.

Release should be done with changing version in package.xml
( change log have to be needed, but ... )

YoheiKakiuchi, 2018-12-10:
@sasabot 
aero-ros-pkgのブランチが増えているのですが、必要なもの以外は増やさないようにしてください。

また、必要性とどうやって使うかをどこか(issueかな)で宣言してください。

sasabot, 2018-12-10:
apologies. I have deleted branches that has already been merged.

fix_lookat_bug is on-going. it is there in case someone needs a fixed version of look at. I will send a pull request once writing some test code around look at and then close once merged. how to use: just compile. it's a bug fix.

add_ps4_teleop_controller. this will be deleted once merged.

0.3.0 is a very old stable version of the seed-solutions package. it is there in case we want to check back in history. users shouldn't use this branch. it's just for reference.

regarding the remaining branches, I'm not aware of these two branches: development, add_readme