seed-solutions/aero-ros-pkg #312: Can't find ros-indigo-pr2-controller-msgs

nagahama-ais, 2018-04-02:
I am installing the aero-ros-pkg and the following error occured.

```
penguin@bizan:~/ros/indigo/src$ sudo apt-get install ros-indigo-pr2-controller-msgs
Reading package lists... Done
Building dependency tree
Reading state information... Done
E: Unable to locate package ros-indigo-pr2-controller-msgs

penguin@bizan:~/ros/indigo/src$ apt-cache search ros-indigo-pr2-controller-msgs
penguin@bizan:~/ros/indigo/src$ 
```

Is there anyone who came up against this problem?

* All other steps succeeded.

sasabot, 2018-04-02:
try
```
sudo apt-get install ros-indigo-pr2-controllers-msgs
```

by the way we are moving on to a new system in a few days. (development branch)
the current master will be moved to a tagged branch and will not be supported in the future.

we recommend new users to use the new system.
more details will be given from @YoheiKakiuchi

nagahama-ais, 2018-04-02:
> sudo apt-get install ros-indigo-pr2-controllers-msgs

Thank you, it worked. 
If you recommend me to wait, I will.

@YoheiKakiuchi sensei, 
Could you announce the finish of updating the system? 

tag: @yamazak

closed 2018-04-11 by YoheiKakiuchi