seed-solutions/aero-ros-pkg #319: Error when update aero-ros-pkg 2.0

hi-kondo, 2018-04-06:
旧バージョンをクリーンインストールしてアップデートしたところ次のエラーが表示されました。
aero_upper/lower共にUSB未接続の状態です。

$roslaunch aero_startup aero_bringup.launch
core service [/rosout] found
process[aero_ros_controller-1]: started with pid [7570]
process[aero_state_publisher-2]: started with pid [7571]
process[cm_spawner-3]: started with pid [7572]
process[aero_hand_controller-4]: started with pid [7574]
[ INFO] [1522979087.877266980]: upper_port: /dev/aero_upper
[ INFO] [1522979087.877365803]: lower_port: /dev/aero_lower
[ INFO] [1522979087.877392117]: cycle: 66.666000 [ms], overlap_scale 2.000000
could not open /dev/aero_upper
Proto: ERROR: invalid header
could not open /dev/aero_lower
Proto: ERROR: invalid header
[ INFO] [1522979089.955186966]: ControllerManager start with 15.000150 Hz
[ERROR] [1522979090.185875080]: Could not load controller 'joint_state_controller' because controller type 'joint_state_controller/JointStateController' does not exist.
[ERROR] [1522979090.186023392]: Use 'rosservice call controller_manager/list_controller_types' to get the available types
[ERROR] [WallTime: 1522979091.188235] Failed to load joint_state_controller
[ERROR] [1522979091.199552757]: Could not load controller 'larm_controller' because controller type 'position_controllers/JointTrajectoryController' does not exist.
[ERROR] [1522979091.199763598]: Use 'rosservice call controller_manager/list_controller_types' to get the available types
[ERROR] [WallTime: 1522979092.201211] Failed to load larm_controller
[ERROR] [1522979092.213778640]: Could not load controller 'rarm_controller' because controller type 'position_controllers/JointTrajectoryController' does not exist.
[ERROR] [1522979092.213939204]: Use 'rosservice call controller_manager/list_controller_types' to get the available types
[ERROR] [WallTime: 1522979093.215978] Failed to load rarm_controller
[ERROR] [1522979093.227650867]: Could not load controller 'head_controller' because controller type 'position_controllers/JointTrajectoryController' does not exist.
[ERROR] [1522979093.227798492]: Use 'rosservice call controller_manager/list_controller_types' to get the available types
[ERROR] [WallTime: 1522979094.229939] Failed to load head_controller
[ERROR] [1522979094.242487071]: Could not load controller 'waist_controller' because controller type 'position_controllers/JointTrajectoryController' does not exist.
[ERROR] [1522979094.242645316]: Use 'rosservice call controller_manager/list_controller_types' to get the available types
[ERROR] [WallTime: 1522979095.245104] Failed to load waist_controller
[ERROR] [1522979095.258727979]: Could not load controller 'lifter_controller' because controller type 'position_controllers/JointTrajectoryController' does not exist.
[ERROR] [1522979095.258898348]: Use 'rosservice call controller_manager/list_controller_types' to get the available types
[ERROR] [WallTime: 1522979096.260407] Failed to load lifter_controller
[ERROR] [1522979096.271988059]: Could not load controller 'lhand_controller' because controller type 'position_controllers/JointTrajectoryController' does not exist.
[ERROR] [1522979096.272125876]: Use 'rosservice call controller_manager/list_controller_types' to get the available types
[ INFO] [1522979096.774266744]: max: 66.955154 [ms], ave: 66.844340 [ms]
[ERROR] [WallTime: 1522979097.274300] Failed to load lhand_controller
[ERROR] [1522979097.286242708]: Could not load controller 'rhand_controller' because controller type 'position_controllers/JointTrajectoryController' does not exist.
[ERROR] [1522979097.286444015]: Use 'rosservice call controller_manager/list_controller_types' to get the available types
[ERROR] [WallTime: 1522979098.288527] Failed to load rhand_controller

joint_state_controller/JointStateController　does not existとあるのでpathかインストール手順が間違っているかかと思われますが。ご教授お願い致します。

sasabot, 2018-04-06:
```
sudo apt-get install ros-indigo-joint-trajectory-controller
```
(change indigo to kinetic if kinetic)

hi-kondo, 2018-04-06:
That's been resolved
sudo apt-get install ros-indigo-joint-trajectory-controller
sudo apt-get install ros-indigo-joint-state-controller

closed 2018-04-06 by hi-kondo