seed-solutions/aero-ros-pkg #327: running gazebo

sasabot, 2018-04-29:
The robot starts collapsing when I use gazebo (see images below). Any advise?
How I installed:
```
cd aero_description
./setup.sh typeFCESy
cd ../aero_ros_controller
catkin bt
source ~/.bashrc
cd ../aero_gazebo
catkin bt
source ~/.bashrc
```

[image]
[image]

outputs
```
$ roslaunch aero_gazebo aero_gazebo.launch

SUMMARY
========

PARAMETERS
 * /amcl/base_frame_id: base_link
 * /amcl/gui_publish_rate: 10.0
 * /amcl/kld_err: 0.05
 * /amcl/kld_z: 0.99
 * /amcl/laser_lambda_short: 0.1
 * /amcl/laser_likelihood_max_dist: 2.0
 * /amcl/laser_max_beams: 60
 * /amcl/laser_max_range: 12.0
 * /amcl/laser_model_type: likelihood_field
 * /amcl/laser_sigma_hit: 0.2
 * /amcl/laser_z_hit: 0.5
 * /amcl/laser_z_max: 0.05
 * /amcl/laser_z_rand: 0.5
 * /amcl/laser_z_short: 0.05
 * /amcl/max_particles: 2000
 * /amcl/min_particles: 500
 * /amcl/odom_alpha1: 0.2
 * /amcl/odom_alpha2: 0.4
 * /amcl/odom_alpha3: 0.2
 * /amcl/odom_alpha4: 0.4
 * /amcl/odom_alpha5: 0.1
 * /amcl/odom_frame_id: odom
 * /amcl/odom_model_type: omni-corrected
 * /amcl/recovery_alpha_fast: 0.0
 * /amcl/recovery_alpha_slow: 0.0
 * /amcl/resample_interval: 1
 * /amcl/transform_tolerance: 1.0
 * /amcl/update_min_a: 0.2
 * /amcl/update_min_d: 0.25
 * /amcl/use_map_topic: True
 * /head_controller/gains/neck_p_joint/d: 0.4
 * /head_controller/gains/neck_p_joint/i: 0.003
 * /head_controller/gains/neck_p_joint/p: 200.0
 * /head_controller/gains/neck_r_joint/d: 0.4
 * /head_controller/gains/neck_r_joint/i: 0.003
 * /head_controller/gains/neck_r_joint/p: 200.0
 * /head_controller/gains/neck_y_joint/d: 0.4
 * /head_controller/gains/neck_y_joint/i: 0.003
 * /head_controller/gains/neck_y_joint/p: 200.0
 * /head_controller/joints: ['neck_y_joint', ...
 * /head_controller/type: effort_controller...
 * /joint_state_controller/publish_rate: 50
 * /joint_state_controller/type: joint_state_contr...
 * /larm_controller/gains/l_elbow_joint/d: 6.0
 * /larm_controller/gains/l_elbow_joint/i: 0.003
 * /larm_controller/gains/l_elbow_joint/p: 1200.0
[... 436 lines cut ...]
[ INFO] [1525043960.891957497, 0.001000000]: Camera Plugin (robotNamespace = /), Info: Using the 'robotNamespace' param: '/'
[INFO] [WallTime: 1525043960.895187] [0.001000] Spawn status: SpawnModel: Successfully spawned model
[ INFO] [1525043960.904114572, 0.001000000]: Camera Plugin (ns = /)  <tf_prefix_>, set to ""
[urdf_spawner-3] process has finished cleanly
log file: /home/t-kasas/.ros/log/4ea79b62-4a69-11e8-8b94-d8d38577bdbb/urdf_spawner-3*.log
[ INFO] [1525043966.191832766, 0.001000000]: Loading gazebo_ros_control plugin
[ INFO] [1525043966.192186461, 0.001000000]: Starting gazebo_ros_control plugin in namespace: /
[ INFO] [1525043966.197347762, 0.001000000]: gazebo_ros_control plugin is waiting for model URDF in parameter [robot_description] on the ROS param server.
[FATAL] [1525043966.368613101, 0.001000000]: No matching hardware interface found for 'hardware_interface/EffortJointInterface
[FATAL] [1525043966.368934232, 0.001000000]: Could not initialize robot simulation interface
[ INFO] [1525043966.426323801, 0.023000000]: waitForService: Service [/gazebo/set_physics_properties] is now available.
[ INFO] [1525043966.463514432, 0.044000000]: Physics dynamic reconfigure ready.
[ INFO] [1525043968.361188000, 1.567000000]: Using plugin "static"
[ INFO] [1525043968.373075979, 1.577000000]: Requesting the map...
[ INFO] [1525043968.620869139, 1.781000000]: Resizing costmap to 800 X 800 at 0.025000 m/pix
[ INFO] [1525043968.744250122, 1.881000000]: Received a 800 X 800 map at 0.025000 m/pix
[ INFO] [1525043968.744366536, 1.881000000]: Subscribing to updates
[ INFO] [1525043968.754424559, 1.887000000]: Using plugin "inflation"
[ INFO] [1525043968.889767332, 1.978000000]: Using plugin "obstacles_laser"
[ INFO] [1525043968.893529835, 1.982000000]:     Subscribed to Topics: laser
[ INFO] [1525043968.924141960, 1.999000000]: Using plugin "inflation"
[FATAL] [1525043968.979026255, 2.048000000]: Failed to create the eband_local_planner/EBandPlannerROS planner, are you sure it is properly registered and that the containing library is built? Exception: According to the loaded plugin descriptions the class eband_local_planner/EBandPlannerROS with base class type nav_core::BaseLocalPlanner does not exist. Declared types are  base_local_planner/TrajectoryPlannerROS dwa_local_planner/DWAPlannerROS
Warning [Publisher.cc:134] Queue limit reached for topic /gazebo/default/pose/local/info, deleting message. This warning is printed only once.
[move_base-9] process has died [pid 7549, exit code 1
```

sasabot, 2018-04-30:
It seems that the problem is caused because the thigh_dummy and shank_dummy are not connected to the leg_knee and leg_base. The model of these two links (lifter_bone_dummy.dae) will rotate around its parent joint causing the robot to collapse. How to fix this? (just adding a dummy joint to the mimic joint does not seem to help)

In addition, the bounding box of the leg_knee and leg_shank seems to be larger than the actual model.
The bounding box of the dummy links does not correspond to the model at all.

[image]
[image]

YoheiKakiuchi, 2018-04-30:
Here is a cause of the problem.
~~~
Warning [ModelDatabase.cc:334] Getting models from[http://gazebosim.org/models/]. This may take a few seconds.

Warning [gazebo.cc:215] Waited 1seconds for namespaces.
Error [gazebo.cc:220] Waited 11 seconds for namespaces. Giving up.
~~~

You can launch gazebo again. 
Then check this line does not exist.
~~~
Warning [ModelDatabase.cc:334] Getting models from[http://gazebosim.org/models/]. This may take a few seconds.
~~~

Another point, EBandPlanner does not exist.
(but it does not relate to this issue)
~~~
[FATAL] [1525043968.979026255, 2.048000000]: Failed to create the eband_local_planner/EBandPlannerROS planner, are you sure it is properly registered and that the containing library is built? Exception: According to the loaded plugin descriptions the class eband_local_planner/EBandPlannerROS with base class type nav_core::BaseLocalPlanner does not exist. Declared types are  base_local_planner/TrajectoryPlannerROS dwa_local_planner/DWAPlannerROS
Warning [Publisher.cc:134] Queue limit reached for topic /gazebo/default/pose/local/info, deleting message. This warning is printed only once.
[move_base-9] process has died [pid 7549, exit code 1
~~~

You can install it with this commnad.
~~~
roscd aero_startup/..
rosdep install --from-paths aero_startup --ignore-src -r
~~~

YoheiKakiuchi, 2018-04-30:
For this problem, you should check your source files. Is there any difference with origin/master ?

> It seems that the problem is caused because the thigh_dummy and shank_dummy are not connected to the leg_knee and leg_base. The model of these two links (lifter_bone_dummy.dae) will rotate around [...]

This picture is current status of origin/master.

[image]

sasabot, 2018-04-30:
There are no diffs but the bounding box is not correct.
```
On branch master
Your branch is up-to-date with 'origin/master'.

Untracked files:
  (use "git add <file>..." to include in what will be committed)

	../../../aero_startup/aero_extra_controllers/AeroHandController.hh
	../../../aero_startup/aero_move_base/AeroBaseController.hh
```
and
```
origin	https://github.com/seed-solutions/aero-ros-pkg (fetch)
origin	https://github.com/seed-solutions/aero-ros-pkg (push)
```
and
```
From https://github.com/seed-solutions/aero-ros-pkg
 * branch            master     -> FETCH_HEAD
Already up-to-date.
```

Tried the rosdep install. Tried launching several times, but still no luck.
output
```
SUMMARY
========

PARAMETERS
 * /amcl/base_frame_id: base_link
 * /amcl/gui_publish_rate: 10.0
 * /amcl/kld_err: 0.05
 * /amcl/kld_z: 0.99
 * /amcl/laser_lambda_short: 0.1
 * /amcl/laser_likelihood_max_dist: 2.0
 * /amcl/laser_max_beams: 60
 * /amcl/laser_max_range: 12.0
 * /amcl/laser_model_type: likelihood_field
 * /amcl/laser_sigma_hit: 0.2
 * /amcl/laser_z_hit: 0.5
 * /amcl/laser_z_max: 0.05
 * /amcl/laser_z_rand: 0.5
 * /amcl/laser_z_short: 0.05
 * /amcl/max_particles: 2000
 * /amcl/min_particles: 500
 * /amcl/odom_alpha1: 0.2
 * /amcl/odom_alpha2: 0.4
 * /amcl/odom_alpha3: 0.2
 * /amcl/odom_alpha4: 0.4
 * /amcl/odom_alpha5: 0.1
 * /amcl/odom_frame_id: odom
 * /amcl/odom_model_type: omni-corrected
 * /amcl/recovery_alpha_fast: 0.0
 * /amcl/recovery_alpha_slow: 0.0
 * /amcl/resample_interval: 1
 * /amcl/transform_tolerance: 1.0
 * /amcl/update_min_a: 0.2
 * /amcl/update_min_d: 0.25
 * /amcl/use_map_topic: True
 * /head_controller/gains/neck_p_joint/d: 0.4
 * /head_controller/gains/neck_p_joint/i: 0.003
 * /head_controller/gains/neck_p_joint/p: 200.0
 * /head_controller/gains/neck_r_joint/d: 0.4
 * /head_controller/gains/neck_r_joint/i: 0.003
 * /head_controller/gains/neck_r_joint/p: 200.0
 * /head_controller/gains/neck_y_joint/d: 0.4
 * /head_controller/gains/neck_y_joint/i: 0.003
 * /head_controller/gains/neck_y_joint/p: 200.0
 * /head_controller/joints: ['neck_y_joint', ...
[... 417 lines cut ...]
[INFO] [WallTime: 1525101997.490114] [0.001000] Spawn status: SpawnModel: Successfully spawned model
[ INFO] [1525101997.496681333, 0.001000000]: Camera Plugin (ns = /)  <tf_prefix_>, set to ""
[urdf_spawner-3] process has finished cleanly
log file: /home/t-kasas/.ros/log/4ea79b62-4a69-11e8-8b94-d8d38577bdbb/urdf_spawner-3*.log
[ INFO] [1525102002.277833051, 0.001000000]: Loading gazebo_ros_control plugin
[ INFO] [1525102002.277933122, 0.001000000]: Starting gazebo_ros_control plugin in namespace: /
[ INFO] [1525102002.278981107, 0.001000000]: gazebo_ros_control plugin is waiting for model URDF in parameter [robot_description] on the ROS param server.
[FATAL] [1525102002.399466762, 0.001000000]: No matching hardware interface found for 'hardware_interface/EffortJointInterface
[FATAL] [1525102002.399722901, 0.001000000]: Could not initialize robot simulation interface
[ INFO] [1525102002.435680941, 0.022000000]: waitForService: Service [/gazebo/set_physics_properties] is now available.
[ INFO] [1525102002.475684218, 0.052000000]: Physics dynamic reconfigure ready.
[ INFO] [1525102004.330845206, 1.544000000]: Using plugin "static"
[ INFO] [1525102004.340490689, 1.554000000]: Requesting the map...
[ INFO] [1525102004.588503908, 1.758000000]: Resizing costmap to 800 X 800 at 0.025000 m/pix
[ INFO] [1525102004.710860771, 1.858000000]: Received a 800 X 800 map at 0.025000 m/pix
[ INFO] [1525102004.710902253, 1.858000000]: Subscribing to updates
[ INFO] [1525102004.720702398, 1.867000000]: Using plugin "inflation"
[ INFO] [1525102004.804751403, 1.926000000]: Using plugin "obstacles_laser"
[ INFO] [1525102004.807348023, 1.926000000]:     Subscribed to Topics: laser
[ INFO] [1525102004.843337091, 1.952000000]: Using plugin "inflation"
Warning [Publisher.cc:134] Queue limit reached for topic /gazebo/default/pose/local/info, deleting message. This warning is printed only once.
[ INFO] [1525102004.899350073, 2.000000000]: Created local_planner eband_local_planner/EBandPlannerROS
[ INFO] [1525102005.229034366, 2.259000000]: Recovery behavior will clear layer obstacles
[ INFO] [1525102005.235509379, 2.265000000]: Recovery behavior will clear layer obstacles
```

sasabot, 2018-04-30:
> This picture is current status of origin/master.

Could you please somehow share the generated model files you are using?

YoheiKakiuchi, 2018-04-30:
You also need to
~~~
roscd aero_startup/..
rosdep install --from-paths aero_gazebo --ignore-src -r
rosdep install --from-paths aero_ros_controller --ignore-src -r
~~~

Here is urdf file.
https://gist.github.com/YoheiKakiuchi/b73305cd777be388a2c7859e2aad8334

sasabot, 2018-04-30:
Is the aero.urdf file the only file required for gazebo? how about aero.urdf.xacro and lifter.urdf.xacro?

YoheiKakiuchi, 2018-04-30:
you can check aero.urdf with aero_description/robots/aero.urdf

sasabot, 2018-04-30:
hmm... so if the aero.urdf is completely the same, should I get the same results in gazebo?
It appears that the generated urdf is the same as the urdf in the link above but I am still encountering the same problems.

Just to make sure, what is the tested version of gazebo and rosdistro?

YoheiKakiuchi, 2018-04-30:
Cause of the problem is here
~~~
[FATAL] [1525102002.399466762, 0.001000000]: No matching hardware interface found for 'hardware_interface/EffortJointInterface
[FATAL] [1525102002.399722901, 0.001000000]: Could not initialize robot simulation interface
~~~
Then, you need some packages.

minimum install process is here
https://github.com/seed-solutions/aero-ros-pkg/blob/master/Dockerfile.kinetic

I checked gazebo with kinetic(docker can run gazebo) and indigo (my laptop)

sasabot, 2018-04-30:
Okay, I will check for packages. In the meantime, I have two questions.

1,
Is ```<hardwareInterface>hardware_interface/EffortJointInterface</hardwareInterface>``` different from ```<hardwareInterface>EffortJointInterface</hardwareInterface>```?
Because the latter seems to work without errors (although that did not solve the problem).

btw. I am using indigo.

2,
I think the common_urdf/gazebo.xacro is also being read? Is this up to date in master?

sasabot, 2018-05-01:
Additional questions.

3, ```[urdf_spawner-3] process has finished cleanly``` should this happen?

4, Are the model parameters correct? Have you ever left the robot standing and leave it for more than one simtime minute? In addition, I have tried removing the dummy links (as these were the links that was obviously moving weirdly). Now, I get the image below. But the model is entering a joint range that is impossible with the real robot (the shank stays correctly).

[image]

6, Why do we need dummy links? If it is for visual purposes, couldn't the thigh and shank .dae contain the dummy bone models?

From what I found, removing the mimic joints did not change the problem in question 4. So my guess is, these problems are not related to the mimic plugin or the PID parameters. However, I do not know what is causing question 4 to happen.

sasabot, 2018-05-01:
I got a similar report from @taichiH
After two minutes in SimTime, the robot bends down. He is using kinetic.

@taichiH
could you please post your screen capture and output log?

YoheiKakiuchi, 2018-05-01:
I found reported error occurred in indigo.
But, I tested in kinetic well.

taichiH, 2018-05-01:
My gazebo robot also starts collapsing like below image.
But, in my case, the robot collapsing once every fifth time.

After collapsing once, I was continuing test launch gazebo model and wait collapsing.
but, at present, the robot doesn't collapsing in the same way.

i'm sorry, I've lost gazebo log when collapsing robot by mistake.

[image]

YoheiKakiuchi, 2018-05-01:
> 1,
> Is <hardwareInterface>hardware_interface/EffortJointInterface</hardwareInterface> different from <hardwareInterface>EffortJointInterface</hardwareInterface>?
> Because the latter seems to work without errors (although that did not solve the problem).

```EffortJointInterface``` is deprecated in kinetic, there are warning messages but working.

> 2,
> I think the common_urdf/gazebo.xacro is also being read? Is this up to date in master?

Yes

> 3, [urdf_spawner-3] process has finished cleanly should this happen?

Yes

> 4

mimic plugin is required to support upper-body with lifter.
We can see messages like that if mimic plugin works correctly.
~~~
Loading mimic plugin
reg: knee_joint -> knee_joint_mimic
offset: 0, multiplier: -1
P: 4000, I: 0.1, D: 80, imax: 500, imin: -500, cmax: 2000, cmin: -2000
~~~

> 6, Why do we need dummy links? If it is for visual purposes, couldn't the thigh and shank .dae contain the dummy bone models?

Dummy links are just for visual. So, geometry for collision is removed.

YoheiKakiuchi, 2018-05-01:
PR https://github.com/seed-solutions/aero-ros-pkg/pull/328 may solve the problem.

sasabot, 2018-05-01:
Hooray!!! It works!
Thank you very much.

but I still see issues with the bounding box (knee and thigh).
also, when you send the below command from [0.0, 0.0], the robot seems to jump.
```
rostopic pub /lifter_controller/commajectory_msgs/JointTrajectory "header:
  seq: 0
  stamp:
    secs: 0
    nsecs: 0
  frame_id: ''
joint_names:
- 'ankle_joint'
- 'knee_joint'
points:
- positions: [1.0, -1.0]
  velocities: [0, 0]
  accelerations: [0, 0]
  effort: [0, 0]
  time_from_start: {secs: 3, nsecs: 0}"
```
can we fix this? or should we wait for the real robot to match the virtual?

YoheiKakiuchi, 2018-05-01:
> but I still see issues with the bounding box (knee and thigh).

This might be a bug of gazebo. It is solved at gazebo7 and kinetic.

> also, when you send the below command from [0.0, 0.0], the robot seems to jump.
> can we fix this? or should we wait for the real robot to match the virtual?

I found it also at kinetic.
It is also a bug or a typical behavior of gazebo.
I think it is hard to fix it. You should use a lifter slower for gazebo.

sasabot, 2018-05-01:
OK thanks.
Then I think we can close this issue for now.

btw. I see a lot of other open issues including those from the old system. maybe we should add tags such as "2.1.0", "2.0.0", "old-system", "old-system-frozen(unsolved)" and close these issues.

closed 2018-05-01 by sasabot