seed-solutions/aero-ros-pkg #335: issues on fingers

sasabot, 2018-05-10:
I have several issues and questions regarding the fingers. First of all, I want to know whether this is a common issue or something that is only happening in my environment (note the environment in parenthesis is the tested distro, doesn't mean it doesn't happen with the other distro). 

1. [issue] Links not finding transform when launching gazebo (kinetic, typeFCESy)
On a clean installed environment, when I ```roslaunch aero_gazebo aero_gazebo.launch``` and ```rosrun rviz rviz``` I get ```No transform from link to [base_link]``` for *l_index_tip_link*, *l_indexend_link*, *l_indexmid_link*, *r_index_tip_link*, *r_indexend_link*, *r_indexmid_link*.
This does not happen with ```roslaunch aero_startup aero_bringup.launch``` and ```rosrun rviz rviz```.

2. [issue] MoveIt planning not considering finger collisions (indigo, typeFCESy)
This might be more of a problem with MoveIt? The fingers in the returned trajectory paths collide with objects in the environment. Anyone else have similar issues? It's really strange because if you check the collision of each trajectory point with ```planning_scene->checkCollision(request, result, robot_state)``` MoveIt correctly returns the links that are in collision.

3. [question] Looking at the urdf, the hand and finger collision are boxes instead of meshes. why?

4. [question] Using AeroMoveitInterface, how to update kinematic_states of fingers?

Anyway, I want to know if issue 1 can be fixed. Issue 2, there might be a problem with the model that is causing the problem? (e.g. the collision are boxes and not meshes?) Thanks in advance.

YoheiKakiuchi, 2018-05-10:
1. I knew what happend. I will fix later. You should use ```roslaunch aero_startup aero_bringup.launch```

2. This problem happens only for indigo?

3. It is for reducing computation load of contact in simulation and collision in MoveIt.
   I think this does not concern problem 2.

sasabot, 2018-05-10:
I believe 2 also happens with kinetic.

sasabot, 2018-05-10:
Regarding 1, the robot collapses like in #327 if I am running ```roslaunch aero_startup aero_bringup.launch```? Maybe it happened by coincidence, I'm not sure.

YoheiKakiuchi, 2018-05-10:
2. I don't understand the situation.
You changed joint angle of fingers ? So, fingers collide with environment.
If fingers was not moved, fingers would not collide.

sasabot, 2018-05-10:
The finger angles doesn't matter. It happens at any angle. The updated code publishes where a collision should be detected during a trajectory (/moveit/collision/contact topic). By running the code multiple times, you should notice that sometimes, the thumb link in the planned trajectory will hit the container (well actually sometimes this happens with the elbow link too, but that one seems more like a small calculation error).

Note, the start and end state does not have any collision. Only the planned trajectory and the joint states in the trajectory are causing collision (so it is not avoiding the obstacle but moving right through the obstacle).

If this is not reproduced in any other machines, then maybe it's just a problem with my ompl yaml settings.

YoheiKakiuchi, 2018-05-10:
I checked planning with this commands.
~~~
roslaunch aero_startup aero_bringup.launch
roslaunch aero_moveit_config real_robot.launch
~~~

[image]

I planned several times. There is no collision finger and environment.
I think RRT returns not so good results and takes much time.
I prefer BKPIECEkConfigDefault (this is default for OMPL??).

here is procedure to add an object and plan
~~~
scene file
chair.scene https://gist.github.com/YoheiKakiuchi/97fcce1c8fef18ad8023d6fd814b0bda
~~~

~~~
In Lower left pannel:
'Scene Object'
  Import From Text => chair.scene
  Publish Scene

In Upper left pannel:
Planning Request
  PlanningGourp => larm_with_waist
  move interactive marker for making goal joint angles

In Lower left pannel:
'Planning'
  Plan
~~~

YoheiKakiuchi, 2018-05-10:
@sasabot 
Please check #337

sasabot, 2018-05-10:
Thanks, I will check. Maybe it is the choice of the algorithm, I'll first check on that.
by the way, are there any updates for issue 1 with the Gazebo?

YoheiKakiuchi, 2018-05-10:
How about #339

sasabot, 2018-05-10:
Changing the algorithm didn't help. So I guess I'll try #337 code.
(pose in picture is goal pose, the transparent is a pose in the path)

[image]

sasabot, 2018-05-10:
By the way, what is the correct way to launch moveit and gazebo? Do I also launch aero_bringup? This somehow creates two robots in rviz ...

sasabot, 2018-05-10:
By the way, when I ```roslaunch aero_gazebo aero_gazebo.launch``` and ```rosrun rviz rviz```, I get this error when I press "execute" after "plan".
```
[ INFO] [1525967346.408394070, 373.866000000]: Execution request received for ExecuteTrajectory action.
[ERROR] [1525967346.408490026, 373.866000000]: Unable to identify any set of controllers that can actuate the specified joints: [ l_elbow_joint l_hand_y_joint l_shoulder_p_joint l_shoulder_r_joint l_shoulder_y_joint l_wrist_p_joint l_wrist_r_joint l_wrist_y_joint r_elbow_joint r_hand_y_joint r_shoulder_p_joint r_shoulder_r_joint r_shoulder_y_joint r_wrist_p_joint r_wrist_r_joint r_wrist_y_joint ]
[ERROR] [1525967346.408527956, 373.866000000]: Known controllers and their joints:

[ INFO] [1525967346.410346377, 373.869000000]: ABORTED: Solution found but controller failed during execution
```

sasabot, 2018-05-10:
#337 couldn't compile on indigo, so I decided to copy your sample and use my AeroMotionPlanningInterface (the non-node version). It seems that your sample works despite node or non-node version. I don't see any collisions with fingers happening with your sample.
Another interesting thing is, I was actually able to successfully solve planning from pose (which I failed to do with my sample).

The difference I see here, is the arm used, the end effector link used, and the planning frame.
I was using the right arm, r_eef_grasp_link, and base_link. Your sample uses left arm, l_hand_link, and /odom (changing /odom to base_link didn't change results).
The other difference is whether planning is solved with goal joints or with end effector pose. And, whether the obstacle is a box or mesh.

Given these differences, I am suspecting the end effector link. I won't check for details at the moment, but I'll try to use the hand_link and see if I encounter similar problems with other manipulation problems.
Anyway, thank you for the sample code!

YoheiKakiuchi, 2018-05-11:
```roslaunch aero_startup aero_bringup.launch``` and ```roslaunch aero_gazebo aero_gazebo.launch``` is mutually exclusive.

If you want to use dynamic simulation, you use gazebo.
If not, you use aero_bringup.launch.

sasabot, 2018-05-11:
OK, thanks. That's exactly what I thought.

Regarding issue 1', it seems that if I send the trajectory results from moveit through sendTrajectory, I can get gazebo to move without any problems (or maybe its because I'm using the right arm).
So basically, there seems to be a way to get everything running.

The last thing I have to check is whether the gazebo collision will work correctly. (I know it won't for indigo, so I'm actually running gazebo on kinetic while everything else on indigo).

sasabot, 2018-05-18:
discussion moved to #337

closed 2018-05-18 by sasabot