seed-solutions/aero-ros-pkg #340: questions and thoughts on current model setup

sasabot, 2018-05-14:
regarding hand
- seed_hand not compatible with typeF_upperbody without yaw joint?
  - seems seed_hand requires a relative offset -0.04 toward z direction?
  - this is a minor issue if all future version comes with a yaw joint
  - the current way to deal with this issue is to create a copy of typeF_upperbody and edit the offset?

regarding move group
- would it be useful to have a group that controls arm + waist(pitch-only, no-yaw)?
  - use cases: dual arm manipulation with mirrored joint angles (one arm copies the other)
  - or are there better ways (standard ways) for doing dual arm manipulation?
  - there is an arm-lifter option but using waist gives more solution range
  - note, arm-waist is not acceptable as this uses waist-yaw (this joint is not symmetric)

sasabot, 2018-05-15:
additional questions and comments:
- hand_yaw.dae in aero_shop but is never referred to from xacro? why is this?
- is the thickness of the yaw joint plate considered? 0.04 is the offset for the hand mesh? or is this the thickness?
- a mesh that shouldn't be attached to the trx_s (the connection to the wrist) is attached in the mesh file

YoheiKakiuchi, 2018-05-16:
@sasabot 
Please describe visually

YoheiKakiuchi, 2018-05-16:
Current status.

There is no problem existing robot type ( typeF and typeFCESy )
hand_yaw + seed_hand => no problem
trx_s (only) => no problem

hand_yaw + trx_s => not good (arm may be longer)
seed_hand (only) => not good (arm may be shorter)

[image]

YoheiKakiuchi, 2018-05-16:
> hand_yaw.dae in aero_shop but is never referred to from xacro? why is this?

hand_yaw.dae should be moved with hand_link. 
Then, in current setting, there is no way to add this mesh to hand_yaw.urdf.
Almost same (opposite) reason for mesh in trx_s.

sasabot, 2018-05-16:
Thank you for the visualization. I see your point.

I upvote for having a dummy_link as we will be having a hand_yaw + trx_s robot soon (but we're not in a real hurry yet).
I think we can have a next minor release around model setups. Including but not limited to: adding dummy_links for shop part compatibility and fixing mesh issues, adding additional joint groups, having a fixed joint version of the wrist and waist, and possibly including the single arm.

by the way, I think we had an issue where the orientation of the forearm was wrong in the model. was this ever fixed? or is it still wrong?