seed-solutions/aero-ros-pkg #341: trx_s hand not working with gazebo

sasabot, 2018-05-18:
tested on kinetic.
the middle index finger is pulled by gravity. how to fix this?
[image]

also, by default midindex is in collision for trx_s. srdf need fixes?

sasabot, 2018-05-18:
never mind. I had to edit my_robot/robots/aero.urdf.xacro for the mimic joints.
but now I'm having trouble opening the fingers. how do I open the trx_s hand on gazebo?

sasabot, 2018-05-18:
I can move the fingers from below command, but if I try to move the hand with /aero_hand_controller service, I get a *Joints don't match* error
```
rostopic pub /rhand_controller/command trajectory_msgs/JointTrajectory "header:
>   seq: 0
>   stamp:
>     secs: 0
>     nsecs: 0
>   frame_id: ''
> joint_names:
> - 'r_indexbase_joint'
> - 'r_thumb_joint'
> points:
> - positions: [-0.1, 0.8]
>   velocities: [0, 0]
>   accelerations: [0, 0]
>   effort: [0, 0]
>   time_from_start: {secs: 3, nsecs: 0}"
```

also I am very confused with these hand controllers
where are they defined?
It seems that the joints used in these controllers have nothing to do with
*my_robot/moveit_config/config/controllers.yaml*
or *my_robot/moveit_config/config/fake_controllers.yaml*

what are these files? where are they used?

I think we need more documents on adding new robot configurations, especially explaining what file influences (or generates) what.

YoheiKakiuchi, 2018-05-18:
You should do ```catkin build aero_ros_controller```

For trs_x , joint_names should be ```[r_thumb_joint]``` ( just one joint )
You can check correct joint_names by ```rostopic echo -n 1 /rhand_controller/state```

sasabot, 2018-05-18:
> You should do catkin build aero_ros_controller

I always rebuild aero_ros_controller every time I change robot configuration, so that should have nothing to do with the problem

> For trs_x , joint_names should be [r_thumb_joint] ( just one joint )

Where is this defined? Do I not use the /aero_hand_controller service for trx_s?
By the way, my controller requires r_indexbase_joint as well.

YoheiKakiuchi, 2018-05-18:
> if the settings are written in robots/controller_settings.yaml and robots/gazebo_controller_settings.yaml, what are the moveit_config/config/controllers.yaml and moveit_config/config/fake_controllers.yaml for?

They are required by moveit (just for 'execute' using move_group ).
fake_controllers.yaml is for demo.launch, not used for real robot.

They are used.
fake_controllers.yaml is automatically generated by moveit_setup_assistant.
controllers.yaml can be converted from robots/controller_settings.yaml, but not motivated to do so.

sasabot, 2018-05-18:
thanks
I have one final question. If only the thumb finger is passed through the controller, how is the trx_s index finger moved?
is there somewhere where I have to define a mimic joint?

YoheiKakiuchi, 2018-05-18:
Settings of mimic joint is in aero_description/my_robot/robots/aer.urdf.xacro
like https://github.com/seed-solutions/aero-ros-pkg/blob/master/aero_description/typeF/robots/aero.urdf.xacro#L47-L205
This is just for gazebo.

no settings in urdf, it seems that fingers are not linear relation ship.

sasabot, 2018-05-18:
great! I think we're almost there.
This is a minor issue but it seems when I send  COMMAND_GRASP, the finger goes to 0 position.
what is the ``` / 4.0``` for in ```L_GRASP()``` of trx_s/controller/AeroHandController.hh?

YoheiKakiuchi, 2018-05-18:
> This is a minor issue but it seems when I send COMMAND_GRASP, the finger goes to 0 position.

Yes. Grasp script does not work in gazebo (it works in seed controller).
So, we have to fix grasp, ungrasp for gazebo.

> what is the / 4.0 for in L_GRASP() of trx_s/controller/AeroHandController.hh?

I don't know why. It is already so at v1.0.0.

closed 2018-05-18 by sasabot