seed-solutions/aero-ros-pkg #344: IK result when top_grasp  with lifter or wholebody

hi-kondo, 2018-05-19:
$ `rosrun aero_samples top_grasp_test` setup typeFCEsy
I set the target position is "obj << 0.6, 0.2, 0.8;
and aero::arm::larm, others defaut value;

When it seems correctly, it moved as below picuture.

[image]

However, sometimes (once a 20 times or less) ik result shows as below picture !!
[image]

When ikrange is upper_body, all ik results in top_grasp_test is correctly.
So I doubt this problem is caused by resolving ik only  range "with lifter".
I'm finding for cause of this problem, but not detecting.
What would you think cause of  this problem?

sasabot, 2018-05-20:
interesting. seems you're not the only one having this problem (happens with ikrange::wholebody).
also, another interesting thing is, ikrange::upperbody will always return the same I.K. results, whereas ikrange::wholebody returns different results everytime you call setFromIK.

since this has never happened with the old system, I would suspect that this has something to do with using the mimic joints.

pictures (expecting above (well not really, the results are kind of random))
[image]

[image]

YoheiKakiuchi, 2018-05-20:
I found that for a target pose which is difficult to find.

I think it is a problem of moveit.
Obviously, IK returns failed pose ( result of FK is different from target ).

sasabot, 2018-05-20:
but why does this specifically happen with ikrange::wholebody? I don't see any problems with ikrange::upperbody.

even with the same goal, ikrange::upperbody returns a valid answer (valid meaning, FK matches target pose) but ikrange::wholebody never returns a valid answer (FK is always different).

by the way, how is the lifter implemented?
I don't see the final conclusion in #283, and I sometimes see #188 happening.

YoheiKakiuchi, 2018-05-29:
please check #351

hi-kondo, 2018-05-29:
I'm using fixed package #351 , it seems no problem for ik_result on lifter.
I will let you know when I have more information.

hi-kondo, 2018-08-23:
I think that it's not particularly a problem in simulation environment and real robot.
So, I'll close this issue.

closed 2018-08-23 by hi-kondo