seed-solutions/aero-ros-pkg #345: waitInterpolation in Gazebo

sasabot, 2018-05-20:
question, is this supported?
apparently, waitInterpolation is returning before action finishes.

measuring time with
```
auto s = ros::Time::now();
robot->sendModelAngles(3000);
robot->waitInterpolation();
ROS_INFO("sleeped for %f", (ros::Time::now() - s).toSec());
```
returns 3.05 seconds, which seems correct if it were the real world.
however, in gazebo, there is a real time factor. therefore, it should actually take more than 3 real world seconds to complete the motion.

sasabot, 2018-05-20:
no wait, the time is correct since ros::Time should be using sim_time.

so there seems to be a different problem where waitInterpolation is returning before the motion completes. is it because in simulation, waitInterpolation just sleeps but the actual speed cannot complete within the given time?
(what is waitInterpolation checking with gazebo?)

YoheiKakiuchi, 2018-05-20:
waitInterpolation returns just after spent commanded time.
It means finished interpolation.
ros_controller limits velocity of joints, then interpolation(moveing joints) doesn't finish within commanded time.
Then we have to command enough time not over velocity limit in urdf.
So, we have to do this https://github.com/seed-solutions/aero-ros-pkg/issues/324

Then it is not gazebo specific problem. ( it is the same real robot and aero_startup/aero_bringup.launch )

sasabot, 2018-05-20:
How about we create a faster model specifically for gazebo?
The real robot has to be slow because of step out, but for simulation, we want the robot to be fast so that we can run multiple simulation tests.

sasabot, 2018-05-21:
tried setting all velocity limit of typeF_upperbody/urdf/.xacro to 100 but still not waiting.

YoheiKakiuchi, 2018-05-21:
You can check /larm_controller/state when desired position stop changing.
( you should choose moving limb_controller )
Please report when start moving desired position,  stop moving, and waitInterpolation returns.