seed-solutions/aero-ros-pkg #349: About difference desired and sctual value of elbow_joint when launch aero_bringup.launch

taichiH, 2018-05-22:
When I launch `roslaunch aero_startup aero_bringup.launch`, robot has to go initial pose (I'm not sure about pose name). 

This is my environment.
aero-ros-pkg branch: master (up-to-date)
Aero model type: typeFCESy

Anyway, I sending  `-2.6379878520965576` to both rarm_elbow and larm_elbow when launch aero_bringup. but I got `-2.444059371948242` as you can see below.

What is the difference of desired and actual value? 

https://github.com/seed-solutions/aero-ros-pkg/blob/e294c4a1615be5c1241ea32de06e8f8ae5fa2605/aero_description/typeFCESy/moveit_config/config/AeroUpperRobot.srdf#L112
Is this value related to this issue? 

Or, is this correct behavior?

```
rostopic echo -n 1 /rarm_controller/state 
header: 
  seq: 22434
  stamp: 
    secs: 1526988631
    nsecs: 347580052
  frame_id: ''
joint_names: [r_shoulder_p_joint, r_shoulder_r_joint, r_shoulder_y_joint, r_elbow_joint, r_wrist_y_joint,
  r_wrist_p_joint, r_wrist_r_joint, r_hand_y_joint]
desired: 
  positions: [0.0, 0.0, 0.0, -2.6379878520965576, 0.0, 0.0, 0.0, -0.00017453292093705386]
  velocities: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
  accelerations: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
  effort: []
  time_from_start: 
    secs: 0
    nsecs:         0
actual: 
  positions: [-0.0, -0.0, -0.0, -2.444059371948242, -0.0, 0.0, 0.0, -0.00017453292093705386]
  velocities: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
  accelerations: []
  effort: []
  time_from_start: 
    secs: 0
    nsecs:         0
error: 
  positions: [0.0, 0.0, 0.0, -0.19392848014831543, 0.0, 0.0, 0.0, 0.0]
  velocities: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
  accelerations: []
  effort: []
  time_from_start: 
    secs: 0
    nsecs:         0
```

YoheiKakiuchi, 2018-05-24:
This is the value for reset_pose used with AeroMoveitInterface .
``` <joint name="l_elbow_joint" value="-2.3562" /> ```

https://github.com/seed-solutions/aero-ros-pkg/blob/v2.1.0/aero_shop/typeF_upperbody/urdf/aero_upper_typef.urdf.xacro#L465

here is limitation for l_elbow_joint (right joint has the same value)

ros controller do not send out of range value.
So, when you launch aero_bringup.launch( ros controller starts ),
move elbow to -2.44246.
And, command for moveing out of range is ignored.

sasabot, 2018-05-24:
is there a reason for the joint limit being restricted than actual (e.g. to avoid collision state or something)? or was this copied from the old codes?
if it was copied, there may be a chance that the limits are the limits of the old upper body (before typeF).

hi-kondo, 2018-05-24:
Although I don't know why elbow joint limit is -2.44246 in urdf, we can set  less than current value of elbow joint limit on specification of hardware. 
Probably, I think this limit value is set because of collision area considered.

sasabot, 2018-06-01:
> Probably, I think this limit value is set because of collision area considered.

Can you explain what you mean by "collision area"?

If you're referring to self-collisions, software collisions can be disabled in srdf. Unless there are hardware collisions, I don't see a reason for limiting these values?
Moreover, when the robot is manipulating in front of a shelf, it is better to tuck the non-manipulating arm as much as possible.