seed-solutions/aero-ros-pkg #353: revise aero_std?

sasabot, 2018-06-02:
possible fixes on next version?

- overall
  - fix time unit to seconds and add warnings for values too large
  - I think it is okay to assume a hand, upper body, lifter, and wheels, but the implementations should not be hard coded (parts not used should have a blank implementation and parameters should be brought from aero_shop. aero_std should be built at setup to avoid troubles).
- lifter
  - remove ```lifter_ik_ ```, implementation should be inside ```setLifter ```
  - consider header implementations for setLifter, getLifter to avoid hard coding on specific type ```bool aero::interface::AeroMoveitInterface::setLifter(double _x, double _z) { return aero::lifter::set(_x, _z); }``` where ```aero::lifter::set``` is defined from a header copied at setup. (to be general and to avoid confusion with joints, input should be aero::Vector3)
- hand related
  - remove ```aero::eef ``` options and add eef conversions instead
  - remove ```sendGraspFast ```; so many problems with current hand, we should forget this until hand is stable
  - consider header implementations for setHand, which allows multiple fingers  ```template<typename... Targs> void aero::interface::AeroMoveitInterface::setHand(aero::arm _arm, Targs&&... _rads) { return aero::hand::set(_arm, {std::forward<Targs>(_rads)...}); }```
  - consider header implementations for getHand, ```double aero::interface::AeroMoveitInterface::getHand(aero::arm _arm, aero::hand::finger _place=aero::hand::finger::thumb) { return aero::hand::get(_arm, _place); }```
- planning related
  - remove ```solveIKSequence```,  ```solveIKOneSequence```,  ```sendSequence```,  ```sendPickIK```,  ```sendPlaceIK```, these should be done using moveit planner
  - add pick poses in header instead? ```aero::Transform aero::poses::pick;```
  - remove ObjectFeatures? markers should use planning scene, and adding sensor etc. should be in MoveitInterface?
- IKSettings.hh
  - add move group with arm + waist pitch?
  - do we need lifter_x, lifter_z in aero::joint? this doesn't work right?
  - aero::joint, joint_map and string_map should be written in a auto-generated file
  - not sure about struct fullarm, there is no real method that can send an arm+hand structure at once (would be nice to have one though)

YoheiKakiuchi, 2018-06-04:
> consider header implementations for setLifter, getLifter to avoid hard coding on specific type bool aero::interface::AeroMoveitInterface::setLifter(double _x, double _z) { return [...]

Currently, origin of arguments (_x, _z) for setLifter is waist position at initial pose.
Then, we need height of initial position.

origin of arguments (_x, _z) for setLifter should be floor level or lifter origin.
Because it does not require initial position and the same value can work with different lifter.

sasabot, 2018-06-04:
another thing is, the function should be abstract anyways, because we have no guarantee that the lifter will always stay two dimensional. (for example, aero-four-legs had a completely different kinematic structure, three dimensional, and was a different implementation)

sasabot, 2018-06-04:
@YoheiKakiuchi 
edited my first comment where I would like to hear your suggestions. thanks.