seed-solutions/aero-ros-pkg #356: joint limits

sasabot, 2018-07-10:
joint limit of r_wrist_r_joint in urdf: 40deg to 75deg
joint limit in csv: 40deg to 80deg
joint limit of actual movement: 20deg to ? (stops at 75 due to urdf limit)
joint limit in excel document: 30deg to 90 deg

@y-shiigi 
What is the correct range?
Also, could you please confirm the correct range for all upper body joints?
Thanks

YoheiKakiuchi, 2018-07-11:
I think each item should be described below.
urdf: limitation of kinematic model, aero_ros_control limits output of command to seed system.
csv: It must be wider than urdf. Because joint angle can not be converted safely if it not contain urdf limitation
document: describing meaning of limitation. it should contain limit in urdf.
(Other limitations are  mechanical limit, mechanical limit + safety margin, actual movement, etc.)

y-shiigi, 2018-07-23:
@sasabot 
The correct data is in documents, and csv data of differential joints were wrong.
Please check https://github.com/seed-solutions/aero-ros-pkg/pull/358.
The reason why csv and urdf are different is as @YoheiKakiuchi said.