seed-solutions/aero-ros-pkg #360: upper body green switch light not blinking after estop

sasabot, 2018-07-26:
second time i see this happen.
the lower body green light blinks but the upper body stays lighted after estop.
cannot re-power the upper body once this happens.

completely powering off the robot fixes the issue but I'm wondering the cause.
also I'd like to know if there is a way to fix this without powering off the robot.

hi-kondo, 2018-07-30:
After you etopped, the cause this happened, what is displayed on LCD of upper robot?  

> cannot re-power the upper body once this happens.

This is , if you release estop and push green light button, upper-body didn't calibrate?

sasabot, 2018-07-30:
> if you release estop and push green light button, upper-body didn't calibrate?

exactly
I don't remember if it was displaying EMG but I'm quite sure it was displaying ROS-MODE.

hi-kondo, 2018-07-30:
I got it.
I'd like to ask one more question.

> the lower body green light blinks but the upper body stays lighted after estop.

Which estop you push when this happened, upper or lower?

sasabot, 2018-07-30:
> Which estop you push when this happened, upper or lower?

I pushed the upper body estop

hi-kondo, 2018-07-30:
This issue happened again, would you please check what is  displayed the LCD of upper.

[image]

or
[image]

> I

 don't remember if it was displaying EMG but I'm quite sure it was displaying ROS-MODE

hi-kondo, 2018-07-30:
> I don't remember if it was displaying EMG but I'm quite sure it was displaying ROS-MODE
I suppose it is displayed as below image from what you say.
[image]

If this happen again, would you please check what is displayed the LCD of upper ?

taichiH, 2019-02-12:
I confirmed the same issue with @sasabot 
In my case, the upper body green light blinks but the lower one stays lighted after estop.