seed-solutions/aero-ros-pkg #363: [Warning] roslaunch aero_startup wheel_with_dummy.launch

hi-kondo, 2018-08-24:
After merged  allow different maps for cost and path #362, 
below warning occured  when launched aero_startup wheel_with_dummy. 

```
seed@seed-ThinkPad-T460s:~/ros/kinetic/src/scenario/src$ roslaunch aero_startup wheel_with_dummy.launch 
... logging to /home/seed/.ros/log/a7e88b58-a790-11e8-9c29-61bcf1f546cc/roslaunch-seed-ThinkPad-T460s-32154.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://seed-ThinkPad-T460s:36813/

SUMMARY
========

PARAMETERS
 * /amcl/base_frame_id: base_link
 * /amcl/gui_publish_rate: 10.0
 * /amcl/kld_err: 0.05
 * /amcl/kld_z: 0.99
 * /amcl/laser_lambda_short: 0.1
 * /amcl/laser_likelihood_max_dist: 2.0
 * /amcl/laser_max_beams: 60
 * /amcl/laser_max_range: 12.0
 * /amcl/laser_model_type: likelihood_field
 * /amcl/laser_sigma_hit: 0.2
 * /amcl/laser_z_hit: 0.5
 * /amcl/laser_z_max: 0.05
 * /amcl/laser_z_rand: 0.5
 * /amcl/laser_z_short: 0.05
 * /amcl/max_particles: 2000
 * /amcl/min_particles: 500
 * /amcl/odom_alpha1: 0.2
 * /amcl/odom_alpha2: 0.4
 * /amcl/odom_alpha3: 0.2
 * /amcl/odom_alpha4: 0.4
 * /amcl/odom_alpha5: 0.1
 * /amcl/odom_frame_id: odom
 * /amcl/odom_model_type: omni-corrected
 * /amcl/recovery_alpha_fast: 0.0
 * /amcl/recovery_alpha_slow: 0.0
 * /amcl/resample_interval: 1
 * /amcl/transform_tolerance: 1.0
 * /amcl/update_min_a: 0.2
 * /amcl/update_min_d: 0.25
 * /amcl/use_map_topic: True
 * /move_base/EBandPlannerROS/differential_drive: False
 * /move_base/EBandPlannerROS/max_acceleration: 0.5
 * /move_base/EBandPlannerROS/max_rotational_acceleration: 1.0
 * /move_base/EBandPlannerROS/max_translational_acceleration: 0.5
 * /move_base/EBandPlannerROS/max_vel_lin: 0.5
 * /move_base/EBandPlannerROS/max_vel_th: 0.5
 * /move_base/EBandPlannerROS/min_vel_lin: 0.1
 * /move_base/EBandPlannerROS/min_vel_th: 0.0
 * /move_base/EBandPlannerROS/xy_goal_tolerance: 0.03
 * /move_base/EBandPlannerROS/yaw_goal_tolerance: 0.05
 * /move_base/TrajectoryPlannerROS/max_rotational_vel: 0.5
 * /move_base/base_global_planner: navfn/NavfnROS
 * /move_base/base_local_planner: eband_local_plann...
 * /move_base/free_thresh: 0.196
 * /move_base/global_costmap/footprint: [[-0.35, -0.25], ...
 * /move_base/global_costmap/footprint_padding: 0.001
 * /move_base/global_costmap/global_frame: map
 * /move_base/global_costmap/inflation/inflation_radius: 0.2
 * /move_base/global_costmap/obstacle_range: 2.5
 * /move_base/global_costmap/obstacles_laser/laser/clearing: True
 * /move_base/global_costmap/obstacles_laser/laser/data_type: LaserScan
 * /move_base/global_costmap/obstacles_laser/laser/inf_is_valid: True
 * /move_base/global_costmap/obstacles_laser/laser/marking: True
 * /move_base/global_costmap/obstacles_laser/laser/topic: scan
 * /move_base/global_costmap/obstacles_laser/observation_sources: laser
 * /move_base/global_costmap/plugins: [{'type': 'costma...
 * /move_base/global_costmap/publish_frequency: 3.0
 * /move_base/global_costmap/raytrace_range: 3.0
 * /move_base/global_costmap/resolution: 0.05
 * /move_base/global_costmap/robot_base_frame: base_link
 * /move_base/global_costmap/rolling_window: False
 * /move_base/global_costmap/static/map_topic: /map
 * /move_base/global_costmap/static/subscribe_to_updates: True
 * /move_base/global_costmap/track_unknown_space: True
 * /move_base/global_costmap/transform_tolerance: 0.5
 * /move_base/global_costmap/update_frequency: 4.0
 * /move_base/image: blank.pgm
 * /move_base/local_costmap/footprint: [[-0.35, -0.25], ...
 * /move_base/local_costmap/footprint_padding: 0.001
 * /move_base/local_costmap/global_frame: odom
 * /move_base/local_costmap/height: 6.0
 * /move_base/local_costmap/inflation/inflation_radius: 0.2
 * /move_base/local_costmap/obstacle_range: 2.5
 * /move_base/local_costmap/obstacles_laser/laser/clearing: True
 * /move_base/local_costmap/obstacles_laser/laser/data_type: LaserScan
 * /move_base/local_costmap/obstacles_laser/laser/inf_is_valid: True
 * /move_base/local_costmap/obstacles_laser/laser/marking: True
 * /move_base/local_costmap/obstacles_laser/laser/topic: scan
 * /move_base/local_costmap/obstacles_laser/observation_sources: laser
 * /move_base/local_costmap/plugins: [{'type': 'costma...
 * /move_base/local_costmap/publish_frequency: 3.0
 * /move_base/local_costmap/raytrace_range: 3.0
 * /move_base/local_costmap/resolution: 0.05
 * /move_base/local_costmap/robot_base_frame: base_link
 * /move_base/local_costmap/rolling_window: True
 * /move_base/local_costmap/static/map_topic: /map
 * /move_base/local_costmap/static/subscribe_to_updates: True
 * /move_base/local_costmap/transform_tolerance: 0.5
 * /move_base/local_costmap/update_frequency: 4.0
 * /move_base/local_costmap/width: 6.0
 * /move_base/negate: 0
 * /move_base/occupied_thresh: 0.65
 * /move_base/origin: [-10.0, -10.0, 0.0]
 * /move_base/resolution: 0.025
 * /rosdistro: kinetic
 * /rosversion: 1.12.13

NODES
  /
    amcl (amcl/amcl)
    dummy_scan (aero_startup/dummy_scan.py)
    map_server (map_server/map_server)
    move_base (move_base/move_base)

ROS_MASTER_URI=http://localhost:11311

process[dummy_scan-1]: started with pid [32241]
process[map_server-2]: started with pid [32242]
process[amcl-3]: started with pid [32245]
process[move_base-4]: started with pid [32251]
[ WARN] [1535110105.623626464]: Timed out waiting for transform from base_link to map to become available before running costmap, tf error: canTransform: target_frame map does not exist.. canTransform returned after 0.100064 timeout was 0.1.
[ WARN] [1535110110.655699028]: Timed out waiting for transform from base_link to map to become available before running costmap, tf error: canTransform: target_frame map does not exist.. canTransform returned after 0.100061 timeout was 0.1.
[ WARN] [1535110115.687811912]: Timed out waiting for transform from base_link to map to become available before running costmap, tf error: canTransform: target_frame map does not exist.. canTransform returned after 0.100986 timeout was 0.1.
```
As a result of this,  topic /map is not published.

So  changed  amcl,launch as code from <remap from="map" to="costmap"/> to     ,  /map is published.
```
<launch>

  <arg name="use_map_topic" default="true"/>
  <arg name="scan_topic" default="scan" />

  <node pkg="amcl" type="amcl" name="amcl">
    <remap from="map" to="costmap"/>
    <param name="use_map_topic" value="$(arg use_map_topic)"/>
    <!-- Publish scans from best pose at a max of 10 Hz -->
    <param name="odom_model_type" value="omni-corrected"/>
    <param name="gui_publish_rate" value="10.0"/>
    <param name="laser_max_beams" value="60"/>
    <param name="laser_max_range" value="12.0"/>
    <param name="min_particles" value="500"/>
    <param name="max_particles" value="2000"/>
    <param name="kld_err" value="0.05"/>
    <param name="kld_z" value="0.99"/>
    <param name="odom_alpha1" value="0.2"/>
    <param name="odom_alpha2" value="0.4"/>
    <!-- translation std dev, m -->
    <param name="odom_alpha3" value="0.2"/>
    <param name="odom_alpha4" value="0.4"/>
    <param name="odom_alpha5" value="0.1"/>
    <param name="laser_z_hit" value="0.5"/>
    <param name="laser_z_short" value="0.05"/>
    <param name="laser_z_max" value="0.05"/>
    <param name="laser_z_rand" value="0.5"/>
    <param name="laser_sigma_hit" value="0.2"/>
    <param name="laser_lambda_short" value="0.1"/>
    <param name="laser_model_type" value="likelihood_field"/>
    <!-- <param name="laser_model_type" value="beam"/> -->
    <param name="laser_likelihood_max_dist" value="2.0"/>
    <param name="update_min_d" value="0.25"/>
    <param name="update_min_a" value="0.2"/>
    <param name="odom_frame_id" value="odom"/>
    <param name="resample_interval" value="1"/>
    <!-- Increase tolerance because the computer can get quite busy -->
    <param name="transform_tolerance" value="1.0"/>
    <param name="recovery_alpha_slow" value="0.0"/>
    <param name="recovery_alpha_fast" value="0.0"/>
    <remap from="scan" to="$(arg scan_topic)"/>    

    <!-- for SEED Mover -->
    <param name="base_frame_id" value="base_link"/>
  </node>

</launch>
```
Would you please check this warning?

sasabot, 2018-09-03:
@hi-kondo 
please check #364

hi-kondo, 2018-09-14:
It seems travis test of aero-ros-controller fails on kinetic.
Is it no problem?

sasabot, 2018-09-14:
I think this is a different issue and it is probably indigo that is failing and not kinetic. The error I see in the log is
```
[ERROR] [1535978034.455391458]: Trying to getState() when no goal is running. You are incorrectly using SimpleActionClient
```
when the controller is launched (and I believe *only* the controller is launched and there are no tests regarding dummy_wheels?)

hi-kondo, 2018-09-14:
> I think this is a different issue and it is probably indigo that is failing and not kinetic. 

I think so, too. But this travis result [travis result](https://travis-ci.org/seed-solutions/aero-ros-pkg) shows kinetic fails.(I also wonder why it fails regarding dummy_wheels.)

sasabot, 2018-09-14:
I think this has nothing to do with dummy_wheels.
Travis is failing on indigo and succeeding on kinetic with travis 222
but it is failing on kinetic and succeeding on indigo with travis 224
Both should be exactly the same code. Summing both results, both indigo and kinetic *should* pass.

I see it as a problem of travis or the aero controller test. Please open a new issue if this is an issue of your concern.

hi-kondo, 2018-09-14:
Thank you for response.
#363 already been resolved, so this issue is closed.

closed 2018-09-14 by hi-kondo