seed-solutions/aero-ros-pkg #384: Lifter shaking in Gazebo

Omegastick, 2018-11-06:
After running any of the aero_samples programs in Gazebo (launching with `roslaunch aero_gazebo aero_gazebo.launch`) the lifter starts shaking. Any ideas why?

[Video here](https://i.imgur.com/9o1n5SF.gifv)

Running ROS Kinetic on Ubuntu 16.04

y-shiigi, 2018-11-13:
Please adjust controller gain of [here](https://github.com/seed-solutions/aero-ros-pkg/blob/master/aero_description/typeF/robots/gazebo_controller_settings.yaml#L73-L74) . For example, P=5, I=0, D=0. 
It may not be a fundamental solution, but the shaking will be stopped.

Omegastick, 2018-11-15:
Thanks, that works great. Would you like me to make a pull request with that change? If so, it's probably best to apply it to the other robots in aero_description too.