seed-solutions/aero-ros-pkg #397: robot joint suddenly not giving feedback of joint angle

sasabot, 2018-12-27:
what happened:
- /joint_states and /larm_controller/state/actual of left shoulder roll keeps returning 0.019... despite position 

left shoulder on rviz
[image]

actual
[image]

why happened:
- no idea, taichH thinks this happens at some timing after doing grasp/ungrasp multiple times

what is known:
- re-running ros controller **does not fix** this problem
- re-running roscore **does not fix** this problem
- re-running robot power button after e-stop **does fix** this problem
- therefore, it is a firmware problem and not a ros problem?

taichiH, 2018-12-27:
[image]

taichiH, 2018-12-27:
This is rosbag file.
https://drive.google.com/file/d/1HZ1eg5wjywlb0WPFwPt7lGMHSKnsde5t/view?usp=sharing

This is one of the result of `rostopic echo /larm_controller/state` 
```
---
header: 
  seq: 2855
  stamp: 
    secs: 1545906102
    nsecs: 756523109
  frame_id: ''
joint_names: [l_shoulder_p_joint, l_shoulder_r_joint, l_shoulder_y_joint, l_elbow_joint, l_wrist_y_joint,
  l_wrist_p_joint, l_wrist_r_joint, l_hand_y_joint]
desired: 
  positions: [0.12854675948619843, 0.40910676248371625, -0.1054178848862648, -1.7235307693481445, 1.1362093687057495, 0.08366498351097107, 0.07585997134447098, 0.5995205640792847]
  velocities: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
  accelerations: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
  effort: []
  time_from_start: 
    secs: 0
    nsecs:         0
actual: 
  positions: [0.12854675948619843, 0.01910676248371601, -0.1054178848862648, -1.7235307693481445, 1.1362093687057495, 0.08366498351097107, 0.07585997134447098, 0.5995205640792847]
  velocities: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
  accelerations: []
  effort: []
  time_from_start: 
    secs: 0
    nsecs:         0
error: 
  positions: [0.0, 0.39000000000000057, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
  velocities: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
  accelerations: []
  effort: []
  time_from_start: 
    secs: 0
    nsecs:         0
```

taichiH, 2018-12-27:
I confirmed this trouble a while ago with waist_y_joint.