seed-solutions/aero-ros-pkg #400: aero on melodic

sasabot, 2019-04-16:
I'd like to know if anyone has succeeded or encountered similar issues.
It seems that the functions around MoveIt crashes with melodic (cannot proceed the setLifter test in fullbody_test). Probably has something to do with the Affine to Isometry update.

Below is a temporary not so splendid solution that avoids the problem but does not solve the problem.
Thanks.
```
$ git clone https://github.com/ros-planning/moveit
$ cd moveit; git fetch origin 0.10.8
$ git checkout -b 0.10.8 refs/tags/0.10.8
$ export ROS_DISTRO=kinetic
$ catkin build
```

YoheiKakiuchi, 2019-04-16:
Tests were passed at that time (March 13?)
https://travis-ci.org/seed-solutions/aero-ros-pkg/builds/505707425

YoheiKakiuchi, 2019-04-18:
This PR https://github.com/ros-planning/moveit/pull/1382 is the cause of this.
(If this PR is removed, all tests pass)

rhaschke, 2019-04-29:
Did you try MoveIt release 1.0.1? This includes https://github.com/ros-planning/moveit/pull/1382.

If you still have issues with MoveIt > 0.10.8, please perform a [git bisect](https://git-scm.com/docs/git-bisect) to narrow down the offending commit.

YoheiKakiuchi, 2019-05-30:
@rhaschke 
This is the first commit that causes the error.
https://github.com/ros-planning/moveit/commit/5bedb7a31db8ab4ac7e474de762e5dfc42ad4d13

The error is something like 'double free'
This commit just does refactoring, I think.
I have no idea what's wrong.

rhaschke, 2019-06-01:
@sasabot, I tried to replicate your issue locally, but you need to provide more information:
What are the exact steps to reproduce the problem? Just calling `rosrun aero_samples fullbody_test_node`, does nothing...

YoheiKakiuchi, 2019-06-01:
@rhaschke
Thanks for testing

We tested it by git bisect with
https://gist.github.com/YoheiKakiuchi/ec0220af9305e90a65ec67d4e2cf6115
https://gist.github.com/YoheiKakiuchi/941d2eff594249da4950e1a2f7f45c1f

The difficult point for building aero-ros-pkg are,

at first you should build aero_description
~~~
catkin build aero_description
~~~
https://gist.github.com/YoheiKakiuchi/941d2eff594249da4950e1a2f7f45c1f#file-dockerfile-melodic-L34

after that, you should ./setup.sh typeF in aero_desctiption
https://gist.github.com/YoheiKakiuchi/941d2eff594249da4950e1a2f7f45c1f#file-dockerfile-melodic-L36-L38
(it is for building indicated(typeF) robot model file, etc. )

then you build aero_samples and run ```rosrun aero_samples fullbody_test_node```

rhaschke, 2019-06-01:
@YoheiKakiuchi, these steps I already managed. However, running `rosrun aero_samples fullbody_test_node`, just outputs the following debug message:
`[INFO] ros.aero_std: start creating robot_interface`

Other than this, I don't get any output or CPU activity. Is this intended?

sasabot, 2019-06-01:
@YoheiKakiuchi @rhaschke Thank you for investigating.
To run the samples, you also need to run a dummy robot controller. ```roslaunch aero_startup aero_bringup.launch ```
after running the dummy controller (keep this launched), and then running ```rosrun aero_samples fullbody_test_node``` the sample should proceed.

rhaschke, 2019-06-01:
Thanks for this hint. Now I the following are reported on the command line:
```
[ERROR] ros.rosconsole_bridge.console_bridge: Semantic description is not specified for the same robot as the URDF
[ INFO] ros.moveit_core.robot_model: Loading robot model 'AeroTypeFCET'...
[ WARN] ros.moveit_ros_planning.kinematics_plugin_loader: Kinematics solver doesn't support #attempts anymore, but only a timeout.
Please remove the parameter '/robot_description_kinematics/rarm/kinematics_solver_attempts' from your configuration.
[ERROR] ros.moveit_kinematics.kdl: Group 'both_arms' is not a chain
[ERROR] ros.moveit_ros_planning.kinematics_plugin_loader: Kinematics solver of type 'kdl_kinematics_plugin/KDLKinematicsPlugin' could not be initialized for group 'both_arms'
[ERROR] ros.moveit_ros_planning: Kinematics solver could not be instantiated for joint group both_arms.
[ERROR] ros.moveit_kinematics.kdl: Group 'both_arms_with_waist' is not a chain
[ERROR] ros.moveit_ros_planning.kinematics_plugin_loader: Kinematics solver of type 'kdl_kinematics_plugin/KDLKinematicsPlugin' could not be initialized for group 'both_arms_with_waist'
[ERROR] ros.moveit_ros_planning: Kinematics solver could not be instantiated for joint group both_arms_with_waist.
[ERROR] ros.moveit_kinematics.kdl: Group 'rarm_with_lifter' is not a chain
[ERROR] ros.moveit_ros_planning.kinematics_plugin_loader: Kinematics solver of type 'kdl_kinematics_plugin/KDLKinematicsPlugin' could not be initialized for group 'rarm_with_lifter'
[ERROR] ros.moveit_ros_planning: Kinematics solver could not be instantiated for joint group rarm_with_lifter.
[ERROR] ros.moveit_kinematics.kdl: Group 'larm_with_lifter' is not a chain
[ERROR] ros.moveit_ros_planning.kinematics_plugin_loader: Kinematics solver of type 'kdl_kinematics_plugin/KDLKinematicsPlugin' could not be initialized for group 'larm_with_lifter'
[ERROR] ros.moveit_ros_planning: Kinematics solver could not be instantiated for joint group larm_with_lifter.
[ERROR] ros.moveit_kinematics.kdl: Group 'upper_body' is not a chain
[ERROR] ros.moveit_ros_planning.kinematics_plugin_loader: Kinematics solver of type 'kdl_kinematics_plugin/KDLKinematicsPlugin' could not be initialized for group 'upper_body'
[ERROR] ros.moveit_ros_planning: Kinematics solver could not be instantiated for joint group upper_body.
[ERROR] ros.moveit_kinematics.kdl: Group 'whole_body' is not a chain
[ERROR] ros.moveit_ros_planning.kinematics_plugin_loader: Kinematics solver of type 'kdl_kinematics_plugin/KDLKinematicsPlugin' could not be initialized for group 'whole_body'
[ERROR] ros.moveit_ros_planning: Kinematics solver could not be instantiated for joint group whole_body.
```
You should fix your .srdf file(s) to handle them.
Finally, there is a
`free(): invalid next size (fast)`
Interestingly, I don't see this issue when running with a memory sanitizer. Looks like a timing/synchronization issue in the `fullbody_test_node` code.

rhaschke, 2019-06-02:
I was able to track down the issue: it's related to the two mimic joints in your "leg". I will file a PR for MoveIt to fix it.

rhaschke, 2019-06-05:
@YoheiKakiuchi, can you confirm that https://github.com/ros-planning/moveit/pull/1490 resolves your issue?

hi-kondo, 2019-07-24:
I checked moveit demo.launch , so error and segfault on melodic.
I tried to control lifter on move_group.
```
[ INFO] [1563963063.195973885]: Constructing new MoveGroup connection for group 'rarm_with_waist' in namespace ''
[ INFO] [1563963064.164067930]: Ready to take commands for planning group rarm_with_waist.
[ INFO] [1563963064.164143969]: Looking around: no
[ INFO] [1563963064.164180162]: Replanning: no
[ INFO] [1563963070.687721504]: Constructing new MoveGroup connection for group 'lifter' in namespace ''
[ INFO] [1563963070.716163254]: Ready to take commands for planning group lifter.
[ INFO] [1563963070.716225071]: Looking around: no
[ INFO] [1563963070.716244467]: Replanning: no

double free or corruption (out)
[rviz_f25f950d6097_3540_3681243212656806726-6] process has died [pid 3602, exit code -6, cmd /opt/ros/melodic/lib/rviz/rviz -d /home/seed/ros/melodic/src/seed_r7_ros_pkg/seed_r7_typef_moveit_config/launch/moveit.rviz __name:=rviz_f25f950d6097_3540_3681243212656806726 __log:=/root/.ros/log/c8ac2112-adfa-11e9-99d6-0242ac110006/rviz_f25f950d6097_3540_3681243212656806726-6.log].
log file: /root/.ros/log/c8ac2112-adfa-11e9-99d6-0242ac110006/rviz_f25f950d6097_3540_3681243212656806726-6*.log
```
[image]
Didn't it fix yet?