seed-solutions/aero-ros-pkg #404: Any idea for control trx-s hand by angle

taichiH, 2019-04-24:
Hi, I would like to adjust grasp by angle.
Now, AeroHandController support only grasp or ungrasp, because of below PR https://github.com/seed-solutions/aero-ros-pkg/pull/355

Is any idea for control grasp angle? @sasabot 
for example, I think, setting actionlib goal (`/lhand_controller/follow_joint_trajectory/goal`) is a simple solution. Is there actionlib client for lhand in aero-ros-pkg?

sasabot, 2019-04-25:
I think grasp by angle is supported. I don't see a reason why it is not.

Could you please report what happens when you try to call ```setHand ``` and ```sendHand ```? (try adding these lines in fullbody_test_node)

Also, make sure you're not seeing errors in your aero_bringup screen.
The number of joints you see from feedback should match the number of joints you see when sending commands. If you see a difference, then something is not set correctly in your robot configuration. (#341 )

What happens if you add indexbase_joints as in [here](https://github.com/seed-solutions/aero-ros-pkg/blob/master/aero_shop/seed_hand/controller/AeroHandController.hh) to [this file](https://github.com/seed-solutions/aero-ros-pkg/blob/master/aero_shop/trx_s/controller/AeroHandController.hh)?

taichiH, 2019-05-14:
Send GraspAngle by below command.
```
rosservice call /aero_hand_controller "{hand: $(python -c 'import aero_startup.srv; print aero_startup.srv.HandControlRequest.HAND_LEFT'), command: $(python -c 'import aero_startup.srv; print aero_startup.srv.HandControlRequest.COMMAND_GRASP_ANGLE'), power: 0, time_sec: 1.0, larm_angle: 0.0, rarm_angle: 0.0}"
```

this is log during sending command (logger level: debug)
```
 5月 15 00:25:53 aeroc jsk.aero.robot.service[21022]: '[DEBUG] [1557847552.817313523] [/aero_hand_controller:ros.roscpp.roscpp_internal]: Accepted connection on socket [7], new socket [14]'
 5月 15 00:25:53 aeroc jsk.aero.robot.service[21022]: '[DEBUG] [1557847552.818048916] [/aero_hand_controller:ros.roscpp]: Adding tcp socket [14] to pollset'
 5月 15 00:25:53 aeroc jsk.aero.robot.service[21022]: '[DEBUG] [1557847552.818126352] [/aero_hand_controller:ros.roscpp.roscpp_internal]: TCPROS received a connection from [133.11.216.162:45664]'
 5月 15 00:25:53 aeroc jsk.aero.robot.service[21022]: '[DEBUG] [1557847552.818234250] [/aero_hand_controller:ros.roscpp.roscpp_internal]: Connection: Creating ServiceClientLink for service [/aero_hand_controller] connected to [callerid=[/rosservice] address=[TCPROS connection on port 57945 to [133.11.216.162:45664 on socket 14]]]'
 5月 15 00:25:53 aeroc jsk.aero.robot.service[21022]: '[DEBUG] [1557847552.818283908] [/aero_hand_controller:ros.roscpp.roscpp_internal]: Service client [/rosservice] wants service [/aero_hand_controller] with md5sum [*]'
 5月 15 00:25:53 aeroc jsk.aero.robot.service[21022]: '[DEBUG] [1557847552.820816904] [/aero_hand_controller:ros.roscpp.roscpp_internal]: Socket [14] received 0/4 bytes, closing'
 5月 15 00:25:53 aeroc jsk.aero.robot.service[21022]: '[DEBUG] [1557847552.820940763] [/aero_hand_controller:ros.roscpp.roscpp_internal]: TCP socket [14] closed'
 5月 15 00:25:53 aeroc jsk.aero.robot.service[21022]: '[DEBUG] [1557847552.820982440] [/aero_hand_controller:ros.roscpp.roscpp_internal]: Connection::drop(0)'
 5月 15 00:25:53 aeroc jsk.aero.robot.service[21022]: '[DEBUG] [1557847552.821014607] [/aero_hand_controller:ros.roscpp.roscpp_internal]: Connection::drop(2)'
 5月 15 00:25:53 aeroc jsk.aero.robot.service[21022]: '[DEBUG] [1557847552.821232737] [/aero_hand_controller:ros.roscpp.roscpp_internal]: Connection::drop(2)'
 5月 15 00:25:53 aeroc jsk.aero.robot.service[21022]: '[DEBUG] [1557847552.862147896] [/aero_hand_controller:ros.aero_ros_controller.actionlib]: Getting status over the wire.'
 5月 15 00:25:53 aeroc jsk.aero.robot.service[21022]: '[DEBUG] [1557847552.862179713] [/aero_hand_controller:ros.aero_ros_controller.actionlib]: Getting status over the wire.'
 5月 15 00:25:53 aeroc jsk.aero.robot.service[21022]: '[DEBUG] [1557847553.043255710] [/aero_hand_controller:ros.roscpp.roscpp_internal]: Accepted connection on socket [7], new socket [14]'
 5月 15 00:25:53 aeroc jsk.aero.robot.service[21022]: '[DEBUG] [1557847553.043729793] [/aero_hand_controller:ros.roscpp]: Adding tcp socket [14] to pollset'
 5月 15 00:25:53 aeroc jsk.aero.robot.service[21022]: '[DEBUG] [1557847553.043771329] [/aero_hand_controller:ros.roscpp.roscpp_internal]: TCPROS received a connection from [133.11.216.162:45678]'
 5月 15 00:25:53 aeroc jsk.aero.robot.service[21022]: '[DEBUG] [1557847553.043829343] [/aero_hand_controller:ros.roscpp.roscpp_internal]: Connection: Creating ServiceClientLink for service [/aero_hand_controller] connected to [callerid=[/rosservice_12400_1557847552820] address=[TCPROS connection on port 57945 to [133.11.216.162:45678 on socket 14]]]'
 5月 15 00:25:53 aeroc jsk.aero.robot.service[21022]: '[DEBUG] [1557847553.043855713] [/aero_hand_controller:ros.roscpp.roscpp_internal]: Service client [/rosservice_12400_1557847552820] wants service [/aero_hand_controller] with md5sum [9c973df02df2e8a99a78e2144605e4bc]'
 5月 15 00:25:53 aeroc jsk.aero.robot.service[21022]: '[DEBUG] [1557847553.049522347] [/aero_hand_controller:ros.aero_ros_controller]: HandControl com: 3, hand 2, pow: 0, time: 1.000000, thre: 0.000000 0.000000, lr_ang: 0.000000 0.000000'
 5月 15 00:25:53 aeroc jsk.aero.robot.service[21022]: '[DEBUG] [1557847553.049555144] [/aero_hand_controller:ros.aero_ros_controller]: Grasp Angle: 2 0.000000 0.000000 1.000000'
 5月 15 00:25:53 aeroc jsk.aero.robot.service[21022]: '[DEBUG] [1557847553.049569327] [/aero_hand_controller:ros.aero_ros_controller]: GraspAngle: sendAngles'
 5月 15 00:25:53 aeroc jsk.aero.robot.service[21022]: '[DEBUG] [1557847553.049611514] [/aero_hand_controller:ros.aero_ros_controller.actionlib]: IN DELETER'
 5月 15 00:25:53 aeroc jsk.aero.robot.service[21022]: '[DEBUG] [1557847553.049634325] [/aero_hand_controller:ros.aero_ros_controller.actionlib]: About to erase CommStateMachine'
 5月 15 00:25:53 aeroc jsk.aero.robot.service[21022]: '[DEBUG] [1557847553.049660227] [/aero_hand_controller:ros.aero_ros_controller.actionlib]: Done erasing CommStateMachine'
 5月 15 00:25:53 aeroc jsk.aero.robot.service[21022]: '[DEBUG] [1557847553.049682'[ERROR] [1557847553.049969912] [/aero_ros_controller:ros.joint_trajectory_controller.rhand_controller]: Joints on incoming goal don't match the controller joints.'
 5月 15 00:25:53 aeroc jsk.aero.robot.service[21022]: 796] [/aero_hand_controller:ros.aero_ros_controller.actionlib]: about to start initGoal()'
 5月 15 00:25:53 aeroc jsk.aero.robot.service[21022]: '[DEBUG] [1557847553.049780064] [/aero_hand_controller:ros.aero_ros_controller.actionlib]: Done with initGoal()'
[... cut ...]
 5月 15 00:25:53 aeroc jsk.aero.robot.service[21022]: '[DEBUG] [1557847553.106113956] [/aero_hand_controller:ros.roscpp.roscpp_internal]: Connection::drop(2)'
 5月 15 00:25:53 aeroc jsk.aero.robot.service[21022]: '[DEBUG] [1557847553.106163371] [/aero_hand_controller:ros.roscpp.roscpp_internal]: Connection::drop(2)'
```

I think this line indicate lhand grasp is correctly done.
```
 5月 15 00:25:53 aeroc jsk.aero.robot.service[21022]: '[DEBUG] [1557847553.050399728] [/aero_hand_controller:ros.aero_ros_controller]: done: lhand'
```
But real robot lhand does not move.

sasabot, 2019-05-15:
1. First of all, is your lhand the trx_s hand you are having trouble with?

2. If so, please post your results on the following commands.
```
rostopic echo -n 1 /lhand_controller/state
```
```
rostopic echo -n 1 /lhand_controller/follow_joint_trajectory/goal
```

3. Also why is your right hand having errors? Is this usual?
> 5月 15 00:25:53 aeroc jsk.aero.robot.service[21022]: '[DEBUG] [1557847553.049682'[ERROR] [1557847553.049969912] [/aero_ros_controller:ros.joint_trajectory_controller.rhand_controller]: Joints on incoming goal don't match the controller joints.'
 5月 15 00:25:53 aeroc jsk.aero.robot.service[21022]: 796] [/aero_hand_controller:ros.aero_ros_controller.actionlib]: about to start initGoal()'

>  5月 15 00:25:53 aeroc jsk.aero.robot.service[21022]: '[DEBUG] [1557847553.050455164] [/aero_hand_controller:ros.aero_ros_controller]: wait (rhand), state = REJECTED '

4. Is this only an issue with the real robot? How about running on rviz without connecting the real robot? (is it a firmware problem or a software problem?)